eclipse-mraa/mraa #207: Erratic PWM behavior

dandyd449, 2015-05-16:
I've been unable to read the current duty cycle percentage.
```
#include "mraa.hpp"
#include <unistd.h>

int main(){
    mraa::Pwm* led;

    led = new mraa::Pwm(3);
    if(led == NULL){
        return MRAA_ERROR_UNSPECIFIED;
    }

    led->enable(true);

    led->write(0.90);
    fprintf(stdout, "%f\n", led->read());

    sleep(2);

    led->write(0.40);
    fprintf(stdout, "%f\n", led->read());

    sleep(2);

    led->enable(false);
    led->~Pwm();
}
```

This should: 
Turn on the led
Set the duty cycle to 90%
Set the duty cycle to 25%
Turn off the led

This does not work. In face it looks as if the led changes to 100% upon exiting. 
However if I set the duty cycle to 1% prior to disabling the led. It will turn off.

There is no log message generated.

cosscat, 2015-05-16:
Hello dandyd449,

What board did you use?
I'm using a Galileo v1 and just compiled your code:
g++ -lmraa pwm_test.cpp
and works for me.

Here is my output:
oot@galileo:~/c++_tests# ./a.out 
0.899998
0.400000

regarding the led->enable(false); i don't think it should turn off the LED. The reason for this is because what pwm does is to refer to the /sys pwm chip as mapped in kernel:
// from mraa/src/pwm/pwm.c after cloning mraa
snprintf(bu, MAX_SIZE, "/sys/class/pwm/pwmchip%d/pwm%d/enable", dev->chipid, dev->pin);
//... then open the file:
int period_f = open(bu, O_RDWR);
//... then write to it:
if (write(period_f, out, length * sizeof(char)) == -1) {
...
What i'm saying is that i tried manually by setting the pwm enable file in /sys fs and i get the same behaviour. To shut down the led, i used the Gpio part of mraa, but from python(it was easier for me):
import mraa as m
>>> gpio = m.Gpio(3);
>>> gpio.dir(m.DIR_OUT)
>>> gpio.write(0)

Costin

tingleby, 2015-05-16:
Based on previous issues filed this is on the edsion yes? There is an outstanding issue with the PWM device is that if disabled it acts as a HIGH pin. 
There is a patch in the works that works when the pin is being set to a duty of 0% the pin is to change mode into a GPIO and be set to LOW

dandyd449, 2015-05-16:
That is correct. This is an edison.

Whats curious is why it seemly works if I set the duty to 1% prior to led->enable(false) and why at 0% it still outputs something. I look forward to this patch for sure. I was very close to abandoning the platform for something else.

Thanks!

tingleby, 2015-06-04:
Sorry for the delay in getting to this.
If you could please upgrade your Edison firmware, the team working on the Kernel drivers seems to have fixed a number of issues regarding PWM, making it behave in a much more predictable manner.
https://software.intel.com/en-us/iot/hardware/edison/downloads
Again if you still have the issue, feel free to re-open and Ill try and see whats wrong :)

dandyd449, 2015-06-05:
I was already on the current version. But just in case I reloaded the firmware.
```
root@edison:~# configure_edison --version
146
root@edison:~# configure_edison --latest-version
146
```

I still have this issue.

dandyd449, 2015-06-05:
I am also unable to reopen the ticket.

tingleby, 2015-06-05:
So the problem is with the Kernel for the PWM, on internal newer versions this seems to be fixed, im going to confirm and hope they release it soon.

closed 2015-06-22 by tingleby