eclipse-mraa/mraa #208: Segmentation Fault

FlippieCoetser, 2015-05-19:
Hi, 

Please see the following code snippet (using Node)
```javascript
var mraa = require('mraa');
var pin36 = new mraa.Gpio(36);
pin36.dir(mraa.DIR_IN);
pin36.isr(mraa.EDGE_RISING, function() { console.log('Trigger')})
```
When triggering the interrupt a few times Node exits with an Segmentation Fault.

Node Version: v0.10.29
MRAA Version: v0.6.2

Any workaround?

Kind Regards,
Flippie

srlmproductions, 2015-05-19:
Related? https://github.com/intel-iot-devkit/mraa/issues/161

arfoll, 2015-05-19:
Can you clarify what platform you are on? @srlmproductions I don't think so.

arfoll, 2015-05-20:
Can you try on latest git master? I had a go with edison and fired interupts at between 1000 & 100000 hz and I can't get it to segfault.

FlippieCoetser, 2015-05-20:
@arfoll I am indeed using Edison.
Note: I loaded the OS image and MRAA in the December time frame which worked well.
It was only after loading the latest version yesterday which broke the interrupt function.
it is also important to mention that the interrupt works for a while before the Segmentation Fault appears.

On the hardware side I am using a capacitive based touched sensor which triggers the input at a fairly high rate. Hence, in the code I also do some denouncing before calling the needed method. 
Unfortunately my C and C++ knowledge is very limited... I am much more familiar Javascript using Node. However, if there is any additional diagnostic you would like to me run, just pass me the instruction and I will do the needful.

I will update the MRAA again today and inform you if the problem persists.

Regards,
Flippie

FlippieCoetser, 2015-05-20:
@arfoll 

I just updated the MRAA library however the problem persists.
Below is a screen shot of the output in the console window.
I manually trigger the capacitive based touched sensor which trigger the debounce code which then logs the 'Input: Click'. Debouncing is happening at a rate of 200 msec. The interrupt is therefore triggered more much more than the 'Input: Click' log displays.  
 
[image]

Happy to run any other diagnostic procedure, just need some instructions.

Thanks for helping!
Regards,
Flippie

arfoll, 2015-05-20:
Ok this is weird, can you give me the output of the following:
```{sh}
$ cat /etc/version
$ node -e "var m = require('mraa'); console.log(m.getVersion())"
```

Then on the miniboard (I'm assuming you're on a miniboard/edison right?) Short Pin 36 (J19-9) to J17-11 (SPI SCK). Then run the following code:

```{js}
var m = require('mraa')
x = new m.Spi(0)
x.frequency(1000)
p36 = new m.Gpio(36)
p36.dir(m.DIR_IN)
p36.isr(m.EDGE_RISING, function() { console.log('Trigger')})
for(var i = 0; i < 100;i++) {
  x.writeByte(0)
}
```

Even changing to 10000 I can't get a segfault, so it looks fairly solid on my end :/. Can you do the same?

FlippieCoetser, 2015-05-20:
MRAA version: V0.6.2 

Unfortunately I am using the GPIO board from Sparkfun which do not have the SPI interface pins.
I have slightly adjusted your code to use another pin as output and switch that on and off.

```javascript

var m = require('mraa')
p46 = new m.Gpio(46)
p46.dir(m.DIR_OUT)
var output = 0

p36 = new m.Gpio(36)
p36.dir(m.DIR_IN)
p36.isr(m.EDGE_RISING, function() { console.log('Trigger')})

for(var i = 0; i < 10000; i++){
 output = output == 0 ? 1:0;
 p46.write(output);
}

```
With this code the ISR triggers correctly. 
What I don't understand is that the older OS Image do work.

I am clueless...

arfoll, 2015-05-20:
ok, I have sparkfun gpio board I can try with as well, but I'm a little confused why you're not seeing the issue with the code you just pasted but you do with your capacitive button. Are you able to compile mraa with debug symbols and then run node via gdb so you can get a stacktrace of the sigsegv? That would give us an idea as to where the error is.

```{sh}
cd ~/mraa/
mkdir build/
cmake -DCMAKE_BUILD_TYPE=DEBUG ..
cd src/javascript
make
make install
```

Then run (hit r in gdb shell to start):
```{sh}
gdb --args node script.js
```

FlippieCoetser, 2015-05-22:
@arfoll Here is the trace you needed.

```
[...]
Input: Click
Input: Click
Input: Click
[Thread 0xb75ffb70 (LWP 4948) exited]

Program received signal SIGSEGV, Segmentation fault.
0xb7bf2cde in v8::Function::Call(v8::Handle<v8::Object>, int, v8::Handle<v8::Value>*) () from usr/lib/libv8.so.3.14.5
(gdb)
```

These were the comamnds I ran to get this output
```
git clone https://github.com/intel-iot-devkit/mraa.git
cd mraa/
mkdir build/
cd build/
cmake -DCMAKE_BUILD_TYPE=DEBUG ..
cd src/
make
make install
```

arfoll, 2015-05-22:
could you get the full backtrace by running `bt` in the (gdb) prompt?

FlippieCoetser, 2015-05-22:
Here you are:

```
[...]
Input: Click
Input: Click
[Thread 0xb7666b70 (LWP 5201) exited]

Program received signal SIGSEGV, Segmentation fault.
0xb7bf2cde in v8::Function::Call(v8::Handle<v8::Object>, int, v8::Handle<v8::Value>*) ()
   from /usr/lib/libv8.so.3.14.5
(gdb) bt
#0  0xb7bf2cde in v8::Function::Call(v8::Handle<v8::Object>, int, v8::Handle<v8::Value>*) ()
   from /usr/lib/libv8.so.3.14.5
#1  0xb76c8aa2 in mraa::Gpio::v8isr(uv_work_s*, int) ()
   from /root/node_modules/mraa/build/Release/mraa.node
#2  0x080ca2c1 in uv__work_done ()
#3  0x080c00a6 in ?? ()
#4  0x080c015c in ?? ()
#5  0x080cd302 in uv__io_poll ()
#6  0x080c09f2 in uv_run ()
#7  0x0806fdb7 in node::Start(int, char**) ()
#8  0x0806259b in main ()

```

FlippieCoetser, 2015-05-22:
I updated to

arfoll, 2015-05-22:
Sorry one last thing - could you past the output of a full `thread apply all backtrace`? I'm curious to see if there is another libuv thread calling v8::Function::Call at the same time.

Another thing you could do would be to build the `jsmutex` branch and see if you can still reproduce.

FlippieCoetser, 2015-05-22:
Downgraded Node again to the latest available via Debian 7.8 package manager `v0.10.38` and here the segfault was also reproducible.

Here is the according gdb output:

```
Input: Click
Input: Click
Input: Click

Program received signal SIGSEGV, Segmentation fault.
0x0832295e in v8::Function::Call(v8::Handle<v8::Object>, int, v8::Handle<v8::Value>*) ()
(gdb) bt
#0  0x0832295e in v8::Function::Call(v8::Handle<v8::Object>, int, v8::Handle<v8::Value>*) ()
#1  0xb7bf0aa2 in mraa::Gpio::v8isr(uv_work_s*, int) ()
   from /root/node_modules/mraa/build/Release/mraa.node
#2  0x08603ab1 in uv__work_done ()
#3  0x085f97e9 in ?? ()
#4  0x085f989c in ?? ()
#5  0x08606b52 in uv__io_poll ()
#6  0x085fa142 in uv_run ()
#7  0x08597887 in node::Start(int, char**) ()
#8  0x081e050b in main ()
(gdb) thread apply all backtrace

Thread 7 (Thread 0xb59ffb70 (LWP 7204)):
#0  0xb7e8ded5 in pthread_cond_wait@@GLIBC_2.3.2 ()
   from /lib/i386-linux-gnu/libpthread.so.0
#1  0x08603508 in uv_cond_wait ()
#2  0x08603864 in ?? ()
#3  0x085f8d51 in ?? ()
#4  0xb7e89954 in start_thread () from /lib/i386-linux-gnu/libpthread.so.0
#5  0xb7e0ac8e in clone () from /lib/i386-linux-gnu/libc.so.6

Thread 6 (Thread 0xb61ffb70 (LWP 7203)):
#0  0xb7e8ded5 in pthread_cond_wait@@GLIBC_2.3.2 ()
   from /lib/i386-linux-gnu/libpthread.so.0
#1  0x08603508 in uv_cond_wait ()
#2  0x08603864 in ?? ()
#3  0x085f8d51 in ?? ()
#4  0xb7e89954 in start_thread () from /lib/i386-linux-gnu/libpthread.so.0
#5  0xb7e0ac8e in clone () from /lib/i386-linux-gnu/libc.so.6

Thread 5 (Thread 0xb69ffb70 (LWP 7202)):
#0  0xb7e8ded5 in pthread_cond_wait@@GLIBC_2.3.2 ()
   from /lib/i386-linux-gnu/libpthread.so.0
#1  0x08603508 in uv_cond_wait ()
#2  0x08603864 in ?? ()
#3  0x085f8d51 in ?? ()
#4  0xb7e89954 in start_thread () from /lib/i386-linux-gnu/libpthread.so.0
#5  0xb7e0ac8e in clone () from /lib/i386-linux-gnu/libc.so.6

Thread 4 (Thread 0xb71ffb70 (LWP 7201)):
#0  0xb7e8ded5 in pthread_cond_wait@@GLIBC_2.3.2 ()
   from /lib/i386-linux-gnu/libpthread.so.0
#1  0x08603508 in uv_cond_wait ()
#2  0x08603864 in ?? ()
#3  0x085f8d51 in ?? ()
#4  0xb7e89954 in start_thread () from /lib/i386-linux-gnu/libpthread.so.0
#5  0xb7e0ac8e in clone () from /lib/i386-linux-gnu/libc.so.6

Thread 3 (Thread 0xb79ffb70 (LWP 7199)):
#0  0xb7dfd42c in poll () from /lib/i386-linux-gnu/libc.so.6
#1  0xb7bde01d in mraa_gpio_interrupt_handler ()
   from /root/node_modules/mraa/build/Release/mraa.node
#2  0xb7e89954 in start_thread () from /lib/i386-linux-gnu/libpthread.so.0
#3  0xb7e0ac8e in clone () from /lib/i386-linux-gnu/libc.so.6

Thread 2 (Thread 0xb7d31b70 (LWP 7198)):
#0  0xb7e900e0 in sem_wait@@GLIBC_2.1 ()
   from /lib/i386-linux-gnu/libpthread.so.0
#1  0x085882e5 in v8::internal::LinuxSemaphore::Wait() ()
#2  0x0849c0a9 in v8::internal::RuntimeProfiler::WaitForSomeIsolateToEnterJS() ()
#3  0x08589628 in v8::internal::SignalSender::Run() ()
#4  0x085884ac in ?? ()
#5  0xb7e89954 in start_thread () from /lib/i386-linux-gnu/libpthread.so.0
#6  0xb7e0ac8e in clone () from /lib/i386-linux-gnu/libc.so.6

Thread 1 (Thread 0xb7d336d0 (LWP 7195)):
#0  0x0832295e in v8::Function::Call(v8::Handle<v8::Object>, int, v8::Handl---Type <return> to continue, or q <return> to quit---
e<v8::Value>*) ()
#1  0xb7bf0aa2 in mraa::Gpio::v8isr(uv_work_s*, int) ()
   from /root/node_modules/mraa/build/Release/mraa.node
#2  0x08603ab1 in uv__work_done ()
#3  0x085f97e9 in ?? ()
#4  0x085f989c in ?? ()
#5  0x08606b52 in uv__io_poll ()
#6  0x085fa142 in uv_run ()
#7  0x08597887 in node::Start(int, char**) ()
#8  0x081e050b in main ()

```

FlippieCoetser, 2015-05-22:
Resolution:
When creating interrupt using `MRAA GPIO v.06.2` ensure you use `Node v.0.12.3` to avoid segment fault. Happy coding!

closed 2015-05-22 by FlippieCoetser