eclipse-mraa/mraa #209: Intel Galileo Gen 2 Python code for Analog read gives error after some time

omkarpujari, 2015-05-20:
Dear Sir , 
I am facing a issue with the intel galileo gen 2 . I am using the mraa library on python 
Here is a sample code which works properly

import mraa
x = mraa.Aio(0)
y = mraa.Aio(1)
....upto mraa.Aio(5)

and then i save it in an array  say 
array = x.read() , y.read() , z.read() , a.read() , b.read()

But when i run a loop say for 100 times the analog read works for some time and suddenly stops 
Please see the attached screenshot for the same.
My whole project is working but i have to use this loop for more than 1000 times and right now it stops even before 100 loops 

[image]

Please reply as early as possible 
Thanks in advance !

alext-mkrs, 2015-05-31:
What mraa version do you have? Have you tried anything else than Python (Node.js or C)?

I'll check this in a moment, but if you could provide the smallest possible program that reproduces the issue, that could be helpful to make sure we're talking about the same thing.

alext-mkrs, 2015-05-31:
Ok, my check will need to wait a day or so because my Galileo Gen2 board needs a console-level intervention to work and I don't have one at hand.

So reproducer program would be really useful until then.

omkarpujari, 2015-06-01:
Please find attached 2 codes for scanning a sensor matrix for one of our
projects .

Both are python codes

In the first one it is properly working

I am scanning 30 sensors for 11 times in a loop and it is perfectly working
.

But in the second code you can see when i make the loop count to 12 it
shows some error after executing 11 times and then stops .

Please find attached the code and please suggest a solution .

I have tried putting a delay after scanning 11 times and then same loop but
the problem persists .

Waiting for your reply.

Ok, my check will need to wait a day or so because my Galileo Gen2 board
needs a console-level intervention to work and I don't have one at hand.

So reproducer program would be really useful until then.

—

import mraa
import os

import json
sensorvalue=[[0 for i in range (30)]for i in range (50)]
holdingvalue=[0 for i in range(11)]
##################################here we change to 12 then it doesnt complete the loop ###

d0 = mraa.Gpio(0) ##for making d0  ON use pin led0
d1 = mraa.Gpio(1) ## d1 for led1
d2 = mraa.Gpio(2)
d3 = mraa.Gpio(3)
d4 = mraa.Gpio(4)
d5 = mraa.Gpio(5)

d6 = mraa.Gpio(6)
d7 = mraa.Gpio(7)

d8 = mraa.Gpio(8)

d0.dir(mraa.DIR_IN)
d1.dir(mraa.DIR_IN)
d2.dir(mraa.DIR_IN)
d3.dir(mraa.DIR_IN)
d4.dir(mraa.DIR_IN)
d5.dir(mraa.DIR_IN)

d6.dir(mraa.DIR_OUT)
d7.dir(mraa.DIR_OUT)

d8.dir(mraa.DIR_IN)

x=[]

############################loop runs for 11 time properly######
##########################loop gives some error when I run for 12 times############3
for i in range (11):
        x = d7.read()
        if (x==1):
                holdingvalue[i]=1
                d6.write(1)   ## motor stop
                d7.write(1)

[... 375 lines cut ...]
                d4.dir(mraa.DIR_OUT)

                d4.write(0)

                a22 = mraa.Aio(0) ## read sensor values
                a23 = mraa.Aio(1)
                a24 = mraa.Aio(2)
                a25 = mraa.Aio(3)
                a26 = mraa.Aio(4)
                a27 = mraa.Aio(5)

        ####sixth scan
                #os.system("echo -n '0'> /sys/class/gpio/gpio18/value ") ## make  gpio     5   direction as output
                #os.system("echo -n '1'> /sys/class/gpio/gpio36/value ") ## make gpio   4   direction as input
                d4.dir(mraa.DIR_IN)
                d5.dir(mraa.DIR_OUT)

                d5.write(0)

                a28 = mraa.Aio(4)
                a29 = mraa.Aio(5)

                d5.write(1)

print(sensorvalue)

alext-mkrs, 2015-06-01:
Ok, thanks, I'll check it out and get back to you.

alext-mkrs, 2015-06-01:
Hmm, any chance you could post this as a gist or check into the repository? Copying the text outside of the GitHub page turns it into a single continuous line, which is totally unreadable and even autopep8 is not able to fix that.

By inspection it looks like you're doing some intertwined GPIO reads/writes with ADC reads and not using any of the read analog values, is this th actual full program?

omkarpujari, 2015-06-02:
Sir,
You may remove everything in my program.
Forget about my program.
If you just write a continuous program to just read analog value and print
its value.
i.e.

while(1)

{
variable=a0.read()
print variable
}

in python in mraa library of intel-iot linux big image on sd card .
Even this loop turns out to give an error after some time.
please find the screenshot attached which gives that error .

are you testing my program on debian or intel-iot image ?
what is the difference between them ? because my one of the friends in
other institute is using the debian image and he said even he is not
getting any error and everything is working properly

alext-mkrs, 2015-06-02:
Thanks, this is much better and easier to test. As I mentioned, I haven't yet been able to test it, just inspected, but generally I was going to use a standard image for that, not an IoT DevKit or Debian. Generally speaking, the driver should be the same in all of them, so there shouldn't be any difference. But we'll see.

In the meanwhile as an experiment you can try adding a small delay into the loop, I suspect it might help.

omkarpujari, 2015-06-02:
I had tried adding delay but in fail.

alext-mkrs, 2015-06-02:
You mean, you've added a delay and it failed in exactly the same way, or the failure was different?

omkarpujari, 2015-06-02:
Same way.
It is showing same error

alext-mkrs, 2015-06-02:
Okay, I've just tested it and I can't reproduce the error you get even if I run it for 10-15 minutes uninterrupted.

To have the level set, here's my configuration:

1. IoT DevKit image version 1.1 downloaded from: http://iotdk.intel.com/images/iot-devkit-latest-mmcblkp0.direct.bz2. I now recall that uClibc-based official image doesn't have MRAA, and neither does my repo based on uClibc image, because there were some problemc building it with uClibc and with older SWIG package that image has. So DevKit image is pretty much the only ready-made option except Debian.
2. MRAA version 0.6.2, installed by running `opkg update && opkg upgrade` after booting the abovementioned image
3. The program I run is listed below
```python
import mraa

print("MRAA version is: " + mraa.getVersion())

a0 = mraa.Aio(0)

while True:
  result = a0.read()
  print result
```

What are your OS image and mraa versions? I suspect the problem might be in one/both of them being outdated.

omkarpujari, 2015-06-04:
Sir,
I am on leave for some time. I will get back after 4 -5 days.

alext-mkrs, 2015-06-04:
Ok, no worries.

rexstjohn, 2015-06-09:
@alext-mkrs is there a way to throw a warning on startup to tell people that their MRAA is out of date? Bet you could avoid a lot of errors like this if there was a default warning if the user's MRAA is older. I have seen several errors submitted which this would solve.

alext-mkrs, 2015-06-09:
Well, technically that could be done, however as long as MRAA is a library and not an end-user application, in my opinion such feature is not really suitable for a library.

It would lead to adding code and additional external dependencies to the library itself, slower application startup for apps using MRAA (due to that Internet polling for newer version) and finally complaints from those who don't read the docs but will notice the connections it's trying to make ("hmm, your library tries to connect to the Internet, that's fishy") and so on.

Plus some part of those devices MRAA-based apps will run on won't have Internet access or will be behind proxy and that won't work anyway.

So sorry if that sounds like I'm turning the idea down :) but I just think the library itself is not the right place to solve this problem. Maybe documentation ("check latest version before submitting a bug" listed everywhere) or lower barrier to updating (though starting from Edison Rel 2.1 the repo configuration is a part of the image, so one just needs to do a `opkg update && opkg install libmraa` to get updated, it doesn't get any easier than that) would be a better solution.

alext-mkrs, 2015-06-20:
@omkarpujari do you still need help with this one? It doesn't honestly looks like a library problem to me, as I've shown below the currently released MRAA version doesn't exhibit this.

omkarpujari, 2015-06-20:
Yes.. I forgot to reply
The image I was using was very old something 0.4...
Now I m using 0.6
But I m not able to connect my board to internet as I am in college campus
and I am not understanding how to login into cyberoam using Intel Galileo.

I want to update this 0.6 image of mraa library to 0.6.2 as you updated
Can you please help me how should I log into cyberoam client using Intel
Galileo ?

alext-mkrs, 2015-06-21:
Cyberoam is I guess a framework used by your college for providing Internet connectivity to students? I haven't worked with that one so I won't be able to advise on the connection, but I can suggest you a way to install packages using offline method.

Just download them manually from the repo using your main computer, then copy them to the board using SD card or SSH-over-RNDIS connectivity with Edison and then install using `opkg install <package_file_name>`. You'll want [this](http://iotdk.intel.com/repos/1.1/intelgalactic/libmraa0_0.6.2_i586.ipk) package.

alext-mkrs, 2015-06-29:
So @omkarpujari have you been able to upgrade the library version since then? Does it work ok?

alext-mkrs, 2015-07-26:
@omkarpujari, ping, so how is it, do you need any more help on this one?

alext-mkrs, 2015-09-03:
@arfoll, I'd propose to close this one as there's no response from the reporter and no one else is chiming in with "me too".

omkarpujari, 2015-09-04:
Dear Sir,
I will get back on Tuesday .
Can I post questions regarding Intel xdk in same thread or M I supposed to
create a new thread ?
I M having a moto g 2nd gen mobile which is not supported in Intel xdk Iot
mobile app . so how should I test my Android application  that I made ?

arfoll, 2015-09-04:
@omkarpujari for XDK or anything that isn't mraa please use the intel IDZ forums, here isn't the right place.

alext-mkrs, 2015-09-20:
@arfoll one more inactive one, I propose closing it.

omkarpujari, 2015-09-20:
Dear sir ,
You can close this .

closed 2015-09-20 by tingleby