eclipse-mraa/mraa #212: "Error: Module did not self-register" with node.js v0.12.4 on Intel Edison

pinwaven, 2015-05-31:
npm install mraa -g

root@edison:~/# node
> var mraa = require('mraa');
Error: Module did not self-register.
    at Error (native)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at repl:1:12
    at REPLServer.defaultEval (repl.js:132:27)
    at bound (domain.js:254:14)
    at REPLServer.runBound [as eval] (domain.js:267:12)
    at REPLServer.<anonymous> (repl.js:279:12)
>

arfoll, 2015-05-31:
Can you past the full output of the compilation of mraa? How did you compile node.js 0.12.4? My guess is that either you compiled mraa before recompiling node or that node 0.10.x is still present on the board and for some reason got chosen.

I've just double checked and mraa 0.7.0 works for me with node 0.12.4

```{sh}
brendan@silver ~ % node --version
v0.12.4
brendan@silver ~ % node
> var mraa = require('mraa')
mraa: FATAL error, failed to initialise platform
undefined
> mraa.getVersion()
'v0.7.0'
```

arfoll, 2015-06-19:
Closing as no update.

closed 2015-06-19 by arfoll