eclipse-mraa/mraa #213: Question about Edison's ADC precision

alext-mkrs, 2015-05-31:
As per our [docs](https://github.com/intel-iot-devkit/mraa/blob/master/docs/edison.md) and [source code](https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/intel_edison_fab_c.c#L1134-L1135) we treat Edison Arduino board's ADC as ADS7955 with "10 bit packed into 12 bit" precision.

At the same time driver sources (even though it calls itself ads7955 there, the precision parameters are for 7951), Edison Arduino board schematic doc and BOM as well as the chip I actually see on my board suggest it's ADS7951, which is "12 bits packed into 16 bits".

So the question is - don't we want to correct that? Even if we want to keep it at 10 bit by default to be Arduino-compatible, at least docs update seems to be in order, to mention actual precision and describe the way to get all 12 bits right away.

I can take care of that, unless there's some reason we have it like that.

alext-mkrs, 2015-06-01:
@arfoll @tingleby any comments on this one?

arfoll, 2015-06-01:
Hey Alex, I had a look at the BOM for the edison arduino breakout (http://www.intel.com/support/edison/sb/CS-035273.htm) and the chip itself is a ADS7951SBRGER which according to TI is a 8 channel 12bit chip as you say (http://www.ti.com/lit/ds/slas605a/slas605a.pdf). I'll try get an official response.

alext-mkrs, 2015-06-01:
Yep, exactly. In addition to that, the very driver used declares it as 12/16 as well.

Thanks, let's see what you find out.

alext-mkrs, 2015-06-03:
Here's by the way a correction of 10-in-16 to 12-in-16 in the driver sources: https://github.com/01org/edison-linux/commit/f4251192d6f20c8895d2c7d23dd621ea05fbe33c

arfoll, 2015-06-05:
Hopefully then newer images will not have this issue. Mraa should be able to deal with it just fine, currently it knows the edison ADC is 12bit and lets you set the res to be 12, obviously if your kernel is 10-in-16 then that'll be a bit pointless for you. Mraa keeps ADC's at 10bits since that's what most people expect and from what I understand quality past 10bits of some 12bit ADCs can be questionable.

closed 2015-06-05 by arfoll

alext-mkrs, 2015-06-05:
The kernel commit was just an additional evidence that it's actually 12-in-16 and that it was corrected there as well. Given the date of the commit, it's highly unlikely anyone uses OS image without this fix - that was 2 official releases ago.

I understand the 10 bit by default setting, though I would still suggest to correct the documentation for the Edison Arduino board - right now we state that
> ADC kernel module will return 12bit number but the ADC itself only has an accuracy of 10bits.

which is not true. Kernel module returns a 16bit number with 12 significant bits.

If you have no objections I'll prepare a PR.

alext-mkrs, 2015-06-05:
Oh and while we are at it (feel free to let me know if there's a better channel for discussing things like that - IRC is a bit quiet :)

In the source code we have three distinct designations of the ADC resolution:
1. `value_bit` field in `struct _aio`
2. `adc_raw` field in `mraa_board_t`
3. `adc_supported` field in `mraa_board_t`

Out of which `adc_supported` isn't really used anywhere in the library code and `value_bit` is hardcoded via `#define` to be 10 + there's a function to set it at run time at user's will (without any boundary checks). In `aio.c` it's only the `raw_bits` and `value_bit` that participate in the result processing.

Could someone please describe what was the idea behind each of those (comments are a bit scarse) and what `adc_supported` is supposed to be used for? My understanding based on the function descriptions in the docs + studying the sources is that `value_bit` should rather be initialized based on `adc_supported` and `adc_raw` should be those "12" and "16" respectively in the "12bit-packed-into-16bit" designation. `value_bit`, when set using that function, should be verified to be not more than `adc_raw` and really not more than `adc_supported` to make sense.

But then the logic we have right now in the code is a bit different, so I wonder what the actual background for such layout is and would be grateful for a clarification.

arfoll, 2015-06-05:
Ah yes - PR away :)

IRC is quiet but I'm usually on there just not talking.r

arfoll, 2015-06-05:
@tingleby can you add some doc around adc_raw and adc_supported? @alext-mkrs is 100% correct that code is a bit messy, there's no bounds checking when calling adc set bits, it should be capped to (probably) adc_raw.

alext-mkrs, 2015-06-06:
@tingleby and feel free to let me know if I can help, I'd be glad to.