eclipse-mraa/mraa #221: UART Failure on Edison

Proclivis, 2015-06-12:
I am having trouble getting the UART to work on Edison, but I have some debug data that might narrow it down. I can't give C code, because I have layered the application over mraa with a FFI of Haskell. But, I have a trace of a working vs. non-working, where with the working case I open the UART with a non-mraa library first, and this fixes it. The traces indicate a difference in initialization.

Details of envinorment:

Ubilinux
Compiled mraa from master
Tracing with strace of application looking for read/write/ioctl

Test case:

The test case does a single byte write at 115200 followed by a read of 6 bytes to a motor controller with a QIK interface. Using a scope, write bytes look the same for the passing and failing case. The failure is at the read, where no bytes come back, and the call to read hangs.

Read code is like this (Haskell) which maps one to one with the mraa lib:

    putStrLn "Open UART"
    uart <- mraaUartInit 0
    print uart
    path <- mraaUartGetDevPath uart
    print path
    putStr "Read Signature: "
    r <- mraaUartSetFlowcontrol uart False False
    r <- mraaUartSetMode uart 8 MraaUartParityNone 1
    r <- mraaUartSetBaudrate uart 115200
    r <- mraaUartWrite uart [0x81] 1
    (r,sig) <- mraaUartRead uart 6
    print sig

Command line looks like:

Open UART                                                                                                                                
0x098c2478                                                                                                                               
"/dev/ttyMFD1"                                                                                                                           
^CRead Signature: [0,0,0,0,0,0] 

Where ^C aborts the application.

Here is a trace of the failure:

write(1, "Open UART\n", 10)             = 10
write(12, "mode1", 5)                   = 5
write(12, "mode1", 5)                   = 5
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_CONTINUE or TCSETSF, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B9600 opost isig icanon echo ...}) = 0
write(1, "0x098c2478\n", 11)            = 11
write(1, "\"/dev/ttyMFD1\"\n", 15)      = 15
ioctl(12, TCXONC, TCIOFF)               = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_CONTINUE or TCSETSF, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_CONTINUE or TCSETSF, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B9600 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_CONTINUE or TCSETSF, {B115200 opost isig icanon echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B115200 opost isig icanon echo ...}) = 0
write(12, "\201", 1)                    = 1
read(12, 0xb710471c, 6)                 = ? ERESTARTSYS (To be restarted)
--- SIGVTALRM (Virtual timer expired) @ 0 (0) ---

The RESTARTS go forever.

Now, I can make a passing case by adding the following code before the mraa code:

-   putStrLn "Pre Open UART"
    sp <- openSerial "/dev/ttyMFD1" defaultSerialSettings { commSpeed = CS115200 }
    closeSerial sp

Once this code runs, the mraa code will work (without the fixing code) until I reboot/powercycle. 

This indicates an initialization problem to me, one I obviously don't understand.

If I trace these two (fixing) lines I see:

write(1, "Pre Open UART\n", 14)         = 14
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B115200 -opost -isig -icanon -echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B115200 -opost -isig -icanon -echo ...}) = 0
ioctl(12, SNDCTL_TMR_START or TCSETS, {B115200 -opost -isig -icanon -echo ...}) = 0
ioctl(12, SNDCTL_TMR_TIMEBASE or TCGETS, {B115200 -opost -isig -icanon -echo ...}) = 0
--- SIGVTALRM (Virtual timer expired) @ 0 (0) ---

Note the virtual timer is just part of the Haskell run time. The line that is different uses:

SNDCTL_TMR_START

Not knowing how the drivers are coded, or any differences there may be between Yocto and Ubilinux, perhaps there is some small missing call required just for Ubilinux?

I can obviously try some small changes to mraa, and can probably add some printf's to rerun and gather data. I would like to get this to work so I can remove the library that I used to make it work. I am trying to decouple from that library and do my whole application layered over mraa. I have SPI and I2C working from Haskell so far without any issues.

arfoll, 2015-06-19:
Sorry for the delay, haskell bindings sound awesome - is your code available for us to see?

I honestly don't really understand the issue, before reading no data maybe just check there is data available?

I don't think there are differences between the poky/YP image and ubilinux since they don't recompile the kernel but I don't really know anything baout ubilinux. I'd recommend getting gdb onto mraa (just compile with debug symbols) and have a look at what's happening?

alext-mkrs, 2015-06-19:
Yeah, ubilinux is just a debian userspace on top of yocto-generated kernel and drivers, so theoretically there should be no difference.

Proclivis, 2015-06-19:
Does anybody know what the extra text line in my log means and if there is any possibility it matters?

alext-mkrs, 2015-06-19:
What exactly do you call an extra text line?

As @arfoll mentioned it would help if you could share your FFI code, otherwise it's a bit hard to say. I personally have analyzed your traces, but as I've never seen FFI at work and from an strace prospective, nothing stands out there.

Proclivis, 2015-06-22:
I will decouple my FFI code and publish on github, then come back and answer in context.

Proclivis, 2015-06-23:
The mraa code is at https://github.com/Proclivis/mraa

The extra line in the trace is:

ioctl(12, SNDCTL_TMR_START or TCSETS, {B115200 -opost -isig -icanon -echo ...}) = 0

And I assume this is doing some setup work for the UART because it contains B115200, a baud rate. I am assuming that since it is the only log line that is different between the working and non working case, that it matters.

The connection of this log line with my code is that the UART library in Haskell is doing some initialization that is different than the mraa initialization. I am not familiar with the UART code, so I can't connect it with a particular line in Haskell at the moment. My hope was that someone that knows the UART driver in the kernel would recognize it this message and understand why this command is significant in making mraa work.

In terms of the Haskell FFI, it is a direct translation from Haskell to C. There is no magic in the middle that should change anything, which is why this is a mystery.

alext-mkrs, 2015-06-29:
Okay, now I see what you mean by saying it's a direct translation. Looks like your FFI code is just a wrapper translating input and output types and functions themselves into the Haskell form without changing anything inside.

One thing I've noticed is that both outputs (working and non-working one) actually have speed setting IOCTL, it's just that in anon-working case it's TCSETSF and in the working one it's TCSETS. These two commands have a distinct difference in behaviour (details [here](http://www.lafn.org/~dave/linux/Serial-Programming-HOWTO.html)) though I have to admit I don't see how that would cause such an effect.

Could you please post two full traces for both failing and passing cases, as gists or in your repo? I'd like to see how a working read looks like.

And by the way, looks like strace might be interferring with the process here - see [this SO question](http://stackoverflow.com/questions/25212866/why-is-strace-showing-erestartsys-for-read), it looks like that ERESTARTSYS in the failing case is caused by strace and without that `read()` would just wait. And waiting (i.e. blocking on no data) is generally expected, because we `open()` UART device file in `O_RDWR` mode without using `O_NONBLOCK`. Is the data actually coming back in both cases, or not coming back in both cases? Or maybe it just happened that in the working case the reply came in and therefore `read()` didn't block?

Let's try to reproduce it once again with full straces and see what's going on in more detail. By the way, output from syslog could be useful too, mraa errors would appear there.

Proclivis, 2015-06-30:
Alex,

The files are:

trace-b.log (hangs) 
trace-g.log (runs)

Found at https://gist.github.com/Proclivis

To understand where things happen, the test code:

runQikTest :: IO ()
runQikTest = do
    putStrLn "Open"
    sp <- Q.openQik
    sig <- signature sp
    putStr "Signature "
    print sig
    id <- getDeviceId sp
    putStr "Id "
    print id
    pwm <- getPwm sp
    putStr "PWM "
    print pam

The first thing that happens is at the line with:

signature sp

which evaluates code code:

signature :: UARTBus -> IO String
signature uart = do
  writeB uart [0x81]
  s <- readB uart 6
  return $ map (chr . fromIntegral) s

And in the log file this corresponds to:

write(11, "\201", 1)  

followed by:

read(11, "2", 6) 

Which is where there is a hang in the trace-b.log

Note the printing is not necessarily sequential in the log. Haskell is running 4 cores and may not flush IO strictly. While ERESTARTSYS may be caused by strace, it might also be caused by the Haskell runtime as well. There are a log of signals being used in the run time.

Mike

alext-mkrs, 2015-07-05:
Thanks for the trace. Well, looking at it the only actual difference I see is that in the "good" case the `ioctl()` calls unset a set of bits in the port mode, while in the "bad" one, they're always set. I don't understand why this is happening because in the MRAA code there's a specific use of the `cfmakeraw()` which should unset those.

The list of arguments is unfortunately truncated in your traces, could you please run strace again, but with `-ff -s 256` arguments, so that we could see the full list?

alext-mkrs, 2015-07-05:
Oh, and `-v` too, please.

Proclivis, 2015-07-09:
Alex,

I added two traces with a -long.log in the name, g/b for good/bad. Plis see link:

https://gist.github.com/Proclivis

Mike

alext-mkrs, 2015-07-19:
Looks like I've found the reason, but let's check first. In my tests it looks like a solution and logically it explains what I see in your logs (difference in flags).

@Proclivis, could you please apply the patch I've just pushed to my fork (https://github.com/alext-mkrs/mraa/commit/f06cc4843f499f4d008df833b5ace4252e8785cf), compile the library and try again? Let me know if you want a binary, I can compile it for you.

Proclivis, 2015-07-20:
> @Proclivis, could you please apply the patch I've just pushed to my fork (alext-mkrs@f06cc48), compile the library and try again? Let me know if you want a binary, I can compile it for you.

I'll try to get to this late today. Or tomorrow at the latest. I can compile it.

Proclivis, 2015-07-21:
Alex,

The patch works. I tested several times with reboots in-between. I no longer need the work around.

If you think there is a better solution that needs testing, I am happy to keep testing solutions. Hopefully this version does not break anybody else’s application.

Mike

alext-mkrs, 2015-07-21:
Thanks for reporting back, I'll submit it later today then. It should not break anything, very much the opposite, because our init code didn't do what it was supposed to.

commit 6d39902a (2015-07-22): uart.c: Fixed UART initialization, now we set and apply defaults

Closes #221.
closed 2015-07-22 by arfoll