eclipse-mraa/mraa #222: Xadow I2C bus

rwaldron, 2015-06-13:
- Xadow board is detected as a 56 pin miniboard, which might imply [I2C bus 1 & 6 are available](https://github.com/intel-iot-devkit/mraa/blob/master/docs/edison.md#intelr-breakout-board) for use.
- A Johnny-Five user reported that the Xadow board uses bus 0 (I don't have one, so I can't confirm)

Similar to the `adcRawBits` and `adcSupportedBits` functions, I think that this could be most easily resolved by introducing a common function that returns a list of available buses. 

JS API: 

```js
// Edison + Arduino board
var mraa = require("mraa");
mraa.i2cBusList(); // [ 6 ]

// Edison + Mini board
var mraa = require("mraa");
mraa.i2cBusList(); // [ 1, 6 ]
```

arfoll, 2015-06-15:
Yes this is correct - Xadow simulates a miniboard so i2c bus 1 & 6 are usable, bus1 is default so that leads 0 to 1 meaning it does work.

The logic of mraa and i2c buses is that there is MAX_BUS and default bus. Anything below <= MAX_BUS goes to default bus unless it's a valid bus number. So 5 will give you bus 1 on xadow but 6 will give you 6. We could return a list of buses but I'm not sure of the value, at the end of the day the user needs to know what bus he's plugged in so needs to look at some kind of platform configuration.

alext-mkrs, 2017-04-08:
Reading through this one, I feel like current mraa's logic is good enough and as @arfoll describes, I also don't see much value in getting a list of buses, as user would anyway not necessarily know which specific I2C controllers they map to anyway.

I propose to close this one, any objections or other opinions?