eclipse-mraa/mraa #228: Return value from mraa_i2c_read_byte_data not sane.

xthunderheartx, 2015-06-20:
Does it make sense to return a negative result from a function that is typed as unsigned?  What if I read a register that legitimately returns 0b11111111?  Like say one of the data registers from the MCP23017's on a Numato I/O extender shield?

/**
 * Read a single byte from i2c context, from designated register
 *
 * @param dev The i2c context
 * @param command The register
 * @return The result of the read or -1 if failed
 
*/

uint8_t mraa_i2c_read_byte_data(mraa_i2c_context dev, const uint8_t command);

And looking at the implementation, it doesn't look like it matches the API doc either.  It appears the returned value will be 0, not -1, if the call to mraa_i2c_smbus_access fails. Also if I read 0b11111111 (0xFF) from the device (a perfectly legitimate value), then the function returns that value which is -1.  If I then test that value, it would (erroneously) indicate failure right?

Or have I just been awake way too long?

uint8_t
mraa_i2c_read_byte_data(mraa_i2c_context dev, uint8_t command)
{
    i2c_smbus_data_t d;

    if (mraa_i2c_smbus_access(dev->fh, I2C_SMBUS_READ, command, I2C_SMBUS_BYTE_DATA, &d) < 0) {
        syslog(LOG_ERR, "i2c: Failed to write");
        return 0;
    }
    return 0x0FF & d.byte;
}

Also applies to mraa_i2c_read_byte and mraa_i2c_read_word_data I would think.

arfoll, 2015-06-20:
Thanks for the catch! You're entirely right, the doc is outdated, we used to have -1. However I don't understand what you say here:

> Also if I read 0b11111111 (0xFF) from the device (a perfectly legitimate value), then the function returns that value which is -1. If I then test that value, it would (erroneously) indicate failure right?

If you read 0xFF, then it'll work just fine. Tbh we don't really need the '0x0FF &' anymore either... That's just 'heritage' from our past use of smbus.c from the roadnarrows i2c lib...

```
-> 0x0FF & 0xFF
 = 255
```

closed 2015-06-20 by arfoll

xthunderheartx, 2015-06-21:
Ok in light of the document change how am I to interpret a result of "0"?  An error?  But it isn't an error.  '0' is a legitimate thing to read from an i2c device isn't it?  For instance we use the Numato GPIO extender shield which has 2 MPC20317's (U2 and U4) providing 28 additional bits of GPIO.  They are arranged in 4 separate read data registers of 8 bits each (two, a&b, on each chip, though bits 0..3 of U4 GPIOA are not used).  Now if I connect dry contact sensors to ground on all 8 bit of one of these registers and they all happen to close and I read the register I will get a '0' (0b00000000).  That is not an error!  That again is a legitimate value to read from the device, just as '-1' (0b11111111) is.

The point is that you cannot return an error code in the same 8 bit value you return 8 bits of data. That is one bit of information too much!  It really doesn't matter how you define the failure result. Both 0x00 and 0xFF (i.e. -1) are legal values for an 8 bit register.

Right?

arfoll, 2015-06-21:
@xthunderheartx agreed, it's not ideal, however I don't want to change types without a bit more thought. Can you detail how you would prefer this to be fixed in #229?

xthunderheartx, 2015-06-21:
Sure.  I'll code it up for ya and you can look at the delta's and pull if you want.  The ideal solution is to change the functions to work like their reciprocal 'write' operations, i.e. pass the result in the return and the data in an uint8/uint16 pointed to in a formal parameter of the function.  Prototypes would look something like:

mraa_result_t mraa_i2c_read_byte_data(mraa_i2c_context dev, uint8_t data, uint8_t command)

This would break users of the current library but maybe appropriate preprocessor measures could be taken to prevent that. I personally hate a bunch of conditional compilation in my code but sometimes it's needed.  Of course the mraa C++ classes that use these function would need to be fixed as well, but I looked at them and seems straight forward enough.  Like I said I'll code it up, then you can smoke it over and decide.

Thanks Brendan!

P.S. I also like the idea of normalizing the return codes around mraa_result_t, which the reciprocal functions use.

arfoll, 2015-06-21:
That sounds like a great plan, in C++ we can just throw exceptions. I've got a few API breaks planned for 1.0 so that would fit right in. I guess your protorype would be more this though right?
```
mraa_result_t mraa_i2c_read_byte_data(mraa_i2c_context dev, uint8_t* data, uint8_t command)
```

Whilst we're breaking the API i'd rather switch the order of command and data, it feels like command should be first. Look forward to your PR I'll be more than happy to review. Cheers!

xthunderheartx, 2015-06-22:
Right-o brother.  Forgot the asterisk and I agree about the order of parameters.  I thought the same thing, and if you're gonna break it, break it the right way ;)  I'll get the PR to you as soon as I have a few minutes.  Thanx!

commit 41d71e6c (2015-06-20): i2c.h: Fix docs that incorectly states -1 is failure

Fixes #228