eclipse-mraa/mraa #230: Segmentation Fault

FlippieCoetser, 2015-06-28:
Hi,

With reference to #208, I was able to reproduce another Segment Fault

Node application using ```MRAA v.06.2``` and ```Node v.0.12.3```: 
- HTTP Server for Website running on port ```192.168.2.147:10000```
- Interrupt configured on ```GPIO(36)``` with ``` DIR_IN, EDGE_RISING ```.

Reproducing Segmentation Fault:
1. Start Application ```node app.js```
2. Trigger external Interrupt > Input: Click

[image]

3. Browse to Website ```http://192.168.2.147:10000/```
4. Trigger external Interrupt again

[image]

Here is the full backtrace
Thread 8
[image]
Thread 7
[image]
Thread 6
[image]
Thread 5
[image]
Thread 4
[image]
Thread 3
[image]
Thread 2
[image]
Thread 1
[image]

Any suggestion?

Regards,
Flippie

arfoll, 2015-06-29:
Unfortunately unless I can reproduce there's not much I can do, can you send me some code which reproduces this on node.js 0.12?

FlippieCoetser, 2015-06-29:
I have stripped down the application to only include three components:
- HTTP Webserver ```WebApp.js```
- Interrupt Configuration ```Core.js```
- Event Management ```Event.js```

The application entry is ```Main.js```
I created a public git repository with the App: https://github.com/FlippieCoetser/mraaTestApp.git

To reproduce:
- ```Node Main.js```
- Trigger the Interrupt ```GPIO(36),  DIR_IN, EDGE_RISING```
- Browse to web app ```<IP Address>:10000```
- Trigger the Interrupt again  ```Segmentation Fault```

Loading files to ```HTTP Webserver``` using either ```FS.readFile or FS.readFileSync``` together with ```MRAA``` cause ```Segmentation Fault.```

Regards,
Flippie

FlippieCoetser, 2015-07-11:
@arfoll I upgraded 

```mraa``` to ```v0.7.3-2-gdf0dbb0``` and
```node``` to ```v0.12.7```

No change, still get the same segmentation fault as above. 
I have attached a github link in the previous post so that you are able to reproduce the segmentation fault on your side.

Any help will be much appreciated!

Regards,
Flippie

arfoll, 2015-07-21:
Are you on ubilinux?

FlippieCoetser, 2015-07-21:
Yes, using the latest Ubilinux. However I did have the same problem before #208 , on a Yockto Image. I suspect the problem is with some conflict between Interrupt and Filesystem features used in Node.js

arfoll, 2015-07-21:
Yes but you said it was fixed by moving to 0.12.x which is fairly likely as the ISR path used with that v8 version is quite a bit better. This seems like a different bug.

FlippieCoetser, 2015-07-21:
Agree, could be a different bug. If there is anything that you want to try out just let me know. Happy to do so!

alext-mkrs, 2015-11-03:
@FlippieCoetser is this still actual? Could you please try with mraa 0.8.1 and whatever Node.js you prefer these days?

alext-mkrs, 2015-11-17:
Closing as no response and potential fix provided. Feel free to add a comment if you need further help.

closed 2015-11-17 by alext-mkrs