eclipse-mraa/mraa #232: I2C bus_id is beyond 8 on DE3815?

yongli3, 2015-07-08:
Please check the below output, on my DE3815(Linux kernel version is 3.19), the I2C bus number range is 0 - 9,  In the mraa_intel_de3815 function, the "for (i = 0; i < 8; i++)", should be for (i = 0; i < 10; i++)" ?

:~# cat /sys/class/dmi/id/board_name
DE3815TYKH

:~# i2cdetect -l
i2c-0   i2c             i915 gmbus ssc                          I2C adapter
i2c-1   i2c             i915 gmbus vga                          I2C adapter
i2c-2   i2c             i915 gmbus panel                        I2C adapter
i2c-3   i2c             i915 gmbus dpc                          I2C adapter
i2c-4   i2c             i915 gmbus dpb                          I2C adapter
i2c-5   i2c             i915 gmbus dpd                          I2C adapter
i2c-6   i2c             DPDDC-B                                 I2C adapter
i2c-7   i2c             DPDDC-C                                 I2C adapter
i2c-8   i2c             i2c-designware-pci                      I2C adapter
i2c-9   i2c             i2c-designware-pci                      I2C adapter

arfoll, 2015-07-08:
It probably should be 9 not 8 since it doesn't need to iterate over i2c-9. Could you test it works with your image and this change? Or provide me with your linux image so I can try.

yongli3, 2015-07-08:
@arfoll, I tested on my DE3815, the 9 is okay. it is better if you can check the open error
fd = open(path, O_RDONLY);
(if fd < 0) continue

tingleby, 2015-07-08:
Right so I think mraa needs to count how many i2c-adapters there are available and scan/walk them.

arfoll, 2015-07-08:
@yongli3 a bunch of fixes are in - can you check? I can either make you a few patches on top of 0.7.2 or just make a 0.7.3 release. What would you prefer?

arfoll, 2015-07-08:
In fact there are many fixes that are interesting, I'll release a 0.7.3

yongli3, 2015-07-09:
Thanks @arfoll! I just found the same issue on Minnowmax. I submitted another issue #233. I think we should use  the same method to detect the I2C bus in minnowmax

closed 2015-07-09 by arfoll