eclipse-mraa/mraa #233: I2C_BUS_DEFAULT is not a fixed number on Minnowmax

yongli3, 2015-07-09:
Please check the below output, the kernel version is 3.19. The default I2C number is 8. I think we need to auto detect the I2c_BUS_DEFAULT in intel_minnow_max.c
cat /sys/class/dmi/id/board_name
MinnowBoard MAX
i2cdetect -l
i2c-0   i2c             i915 gmbus ssc                          I2C adapter
i2c-1   i2c             i915 gmbus vga                          I2C adapter
i2c-2   i2c             i915 gmbus panel                        I2C adapter
i2c-3   i2c             i915 gmbus dpc                          I2C adapter
i2c-4   i2c             i915 gmbus dpb                          I2C adapter
i2c-5   i2c             i915 gmbus dpd                          I2C adapter
i2c-6   i2c             DPDDC-B                                 I2C adapter
i2c-7   smbus           SMBus I801 adapter at 2000              SMBus adapter
i2c-8   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-9   i2c             Synopsys DesignWare I2C adapter         I2C adapter

arfoll, 2015-07-09:
Could you give me the result of:
`cat /sys/class/i2c-dev/i2c-8/dev` and `cat /sys/class/i2c-dev/i2c-9/dev`

I think I have a decent solution to deal with this and de3815 issues in a slightly more re-usable way.

yongli3, 2015-07-10:
Thanks @arfoll:
i2cdetect -l
i2c-0   i2c             i915 gmbus ssc                          I2C adapter
i2c-1   i2c             i915 gmbus vga                          I2C adapter
i2c-2   i2c             i915 gmbus panel                        I2C adapter
i2c-3   i2c             i915 gmbus dpc                          I2C adapter
i2c-4   i2c             i915 gmbus dpb                          I2C adapter
i2c-5   i2c             i915 gmbus dpd                          I2C adapter
i2c-6   i2c             DPDDC-B                                 I2C adapter
i2c-7   smbus           SMBus I801 adapter at 2000              SMBus adapter
i2c-8   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-9   i2c             Synopsys DesignWare I2C adapter         I2C adapter
:~# cat /sys/class/i2c-dev/i2c-8/dev
89:8
:~# cat /sys/class/i2c-dev/i2c-9/dev
89:9

arfoll, 2015-07-10:
Fixed with 66e6d8adcddb7211cdcf54d0c81b88ca74e5aa4b

closed 2015-07-10 by arfoll