eclipse-mraa/mraa #234: Add GB-BXBT-3825 platform support ?

yongli3, 2015-07-09:
http://b2b.gigabyte.com/products/product-page.aspx?pid=5269#ov
It is very similar with DE3815, is there any plan to support it ?

srlmproductions, 2015-07-09:
What exactly do you want to use MRAA on that platform for? As far as I can tell it doesn't have any low level I/O.

yongli3, 2015-07-09:
I want to test the I2C functions using this platform. I found there is a GPIO connector on the board, and SMB_CLK/SMB_DATA

arfoll, 2015-07-09:
There's no plans to support it since I don't have one :) However I'd definately like to support it. Is it the same Custom Solutions Connector than on the DE3815? Do you have one I could get access to?

yongli3, 2015-07-10:
Thanks @arfoll, let me check if I find the device