eclipse-mraa/mraa #241: UART C

seamus1982, 2015-07-18:
Hi,

Apologies, for leaving this message here. I was wondering if here are any example of the UART being used for read and write in C. The example shows the UART context being created, however, nothing is ever done with the context. Are there any examples online? 

Any help would be really appreciated as I am use to programming c# which is obviously not nearly as low level. 

Best regards, 
     Seamus

alext-mkrs, 2015-07-19:
Check out the C++ example, it has a full cycle except reading, is that good enough for you?

arfoll, 2015-07-19:
I've modified the example to perform a write, it's not as complete as the C++ example but shows the basics. Is that enough for you? See e4810262273c9eab4fcddbfeefbe82aba344e

seamus1982, 2015-07-20:
Thanks very much for your help. That is great. Thanks again!

closed 2015-07-20 by seamus1982