eclipse-mraa/mraa #242: Trouble compiling MRAA on Ubilinux with GCC 4.9

madarax64, 2015-07-20:
Hi,
I was trying to compile MRAA on the Ubilinux distribution which is based off of Debian. After being unable to git clone the repo for some reason, I downloaded and unzipped a tarball to go ahead.
 After installing the dependencies (gcc, swig, python2.7, etc etc) I then ran make. Here's a screenshot of what I get:
[image]

Manually examining the offending file and line, I came to find out that if I simply changed the line from;
[code] if (percentage > 1.0f) [/code]
to:
[code] if (percentage > 1) [/code]

Seemed to solve the issue. However, while compiling the examples, this would then happen:

[image]

My Cmake configuration was created using: cmake .. -DBUILDSWIGNODE=OFF -DBUILDSWIGPYTHON=OFF and the last line of gcc -v gives:

...
gcc version 4.9.2 (Debian 4.9.2-10)

I'm not sure, but this seems to be some sort of compiler issue?

srlmproductions, 2015-07-20:
For me, building Node 0.12 on Ubilinux from source worked best, and then mraa seemed to work without any problems.*

This thread has a few notes: https://github.com/intel-iot-devkit/mraa/issues/148

* Ok, there's consistent segfaults. But I'm not sure where to go with that.

alext-mkrs, 2015-07-21:
This is apparently a problem with the compiler caused by the fact that ubilinux userspace is not compiled for Edison and just taken as-is from the Debian repo. There are a few quirks related to that and this looks like one of those. Check out the ubilinux thread in the Edison community.

So this doesn't look like something that can be fixed in mraa.

arfoll, 2015-07-21:
@madarax64 not much I can do about this, just don't use ubilinux. Sorry!
@srlmproductions likely your segfaults are to do with ubilinux. I've not been able to reproduce any of them :(
@alext-mkrs can you post a link to the thread?

closed 2015-07-21 by arfoll

srlmproductions, 2015-07-21:
Naw, I'm not really disagreeing that running mraa on Ubilinux is the cause of the segfaults. But the package manager is a strong motivator not to switch to Yocto. Anyway, I'm not sure how to track down the segfaults at this point.

arfoll, 2015-07-21:
@srlmproductions I understand why people want/use ubilinux, however it's just not done the correct way, it's a dodgy hack and there's nothing I can do really except point my finger :)

You might actually have more luck using precompiled ipk packages (just use `ar vx` and then decompress data.tar.gz on /). You might get more luck that way. IPK packages are super similar to DEB so you might even be able to get `dpkg` to install it for you :)

alext-mkrs, 2015-07-21:
I can't recall which exact thread was that, but I remeber such messages at the very beginning of the Ubilinux. By the way, it wasn't mentioned whether we are talking about Galileo or Edison here - on Galileo chances hitting that were much higher due to slightly different instruction set. On Edison I wouldn't expect that to happen, but probably GCC just got compiled with something specific which gets outside of the supported set.

Main Edison thread is (was) here: https://communities.intel.com/thread/56092 and the Galileo one I can't find any more.

@madarax64, try asking David_J_Hunt in the thread mentioned, David is in a better position to help with this one. I haven't heard of such problems for a while, so I was under impression everything was ironed out - so maybe that's just a single case that slipped through and Emutex folks will be willing to help you.

madarax64, 2015-07-21:
Thanks everyone for the input! 
@alext-mkrs and @arfoll Thanks alot guys, much obliged. @alext-mkrs Its an Intel Galileo Gen 1 specifically. I'll try to reach him on the Intel Communities forum, thanks for the tip
@srlmproductions I'm completely with you regarding that package manager. If Yocto and opkg had a similar breadth of readily-available packages Ubilinux would be less attractive....but it is what is is :-)

alext-mkrs, 2015-07-22:
In case of Galileo yeah, that's unsurprising, due to the fact that Debian is not compiled natively for Quark and they're taking it as-is for Ubilinux. Hopefully Emutex will help with that one.

As far as tje package mansger is concerned - tjere is a repo I'm supporting for Edison and Galileo, which is not a competitor to Debian package-number-wise, but each has several thousand packages. Maybe it will be of help for you, check it at repo.opkg.net/edison/repo, config instructions are at alextgalileo.altervista.org.

madarax64, 2015-07-25:
Hi @alext-mkrs ,
Fingers crossed on Emutex then :-)
Okay, I'll check it out. Thanks for the tip.

Cheers!