eclipse-mraa/mraa #244: mraa_uart_init(1) on galileo returns NULL and syslogs "uart: platform has only 1"

xthunderheartx, 2015-07-22:
Galileo has 2 UARTs, both of which can be exposed on the Arduino header, ttys0 AND ttys1.  Yes, ttys1 normally has the console getty attached but we need the UART for something more important.  The documentation for GEN2 clearly states that UART 0 can be exposed on Arduino pins 0 and 1 and that UART1 can be exposed on Arduino pins 2 and 3.

xthunderheartx, 2015-07-22:
Would it be as simple as changing mraa_intel_galileo_gen2_uart_init_pre(int) to include the pins for TTYS1 which according to this link:

 http://www.emutexlabs.com/component/content/article?id=203:getting-started-with-intel-galileo-gen-2

are gpio_34 (RX) and gpio_16 (TX)?  Something like this maybe:

mraa_result_t
mraa_intel_galileo_gen2_uart_init_pre(int index)
{
    mraa_gpio_context io0_output = mraa_gpio_init_raw(32);
    if (io0_output == NULL) {
        return MRAA_ERROR_INVALID_RESOURCE;
    }
    mraa_gpio_context io1_output = mraa_gpio_init_raw(28);
    if (io1_output == NULL) {
        mraa_gpio_close(io0_output);
        return MRAA_ERROR_INVALID_RESOURCE;
    }

    mraa_gpio_context io2_output = mraa_gpio_init_raw(34);
    if (io2_output == NULL) {
       mraa_gpio_close(io0_output);
       mraa_gpio_close(io1_output);
       return MRAA_ERROR_INVALID_RESOURCE;
    }
    mraa_gpio_context io3_output = mraa_gpio_init_raw(16);
    if (io3_output == NULL) {
        mraa_gpio_close(io0_output);
        mraa_gpio_close(io1_output);
        mraa_gpio_close(io2_output);
        return MRAA_ERROR_INVALID_RESOURCE;
    }

    int status = 0;
    status += mraa_gpio_dir(io0_output, MRAA_GPIO_OUT);
    status += mraa_gpio_dir(io1_output, MRAA_GPIO_OUT);

    status += mraa_gpio_write(io0_output, 1);
    status += mraa_gpio_write(io1_output, 0);

    status += mraa_gpio_dir(io2_output, MRAA_GPIO_OUT);
    status += mraa_gpio_dir(io3_output, MRAA_GPIO_OUT);

    status += mraa_gpio_write(io2_output, 1);
    status += mraa_gpio_write(io3_output, 0);

    mraa_gpio_close(io0_output);
    mraa_gpio_close(io1_output);
    mraa_gpio_close(io2_output);
    mraa_gpio_close(io3_output);

    if (status > 0) {
        return MRAA_ERROR_UNSPECIFIED;
    }
    return MRAA_SUCCESS;
}

and then changing mraa_intel_galileo_gen2() like this:

.
.
.   
    b->uart_dev_count = 2;
    b->def_uart_dev = 0;
    b->uart_dev[0].rx = 0;
    b->uart_dev[0].tx = 1;
    b->uart_dev[0].device_path = "/dev/ttyS0";
    b->uart_dev[1].rx = -1;
    b->uart_dev[1].tx = -1;
    b->uart_dev[1].device_path = "/dev/ttyS1";
.
.
.

 to:

    b->uart_dev_count = 2;
    b->def_uart_dev = 0;
    b->uart_dev[0].rx = 0;
    b->uart_dev[0].tx = 1;
    b->uart_dev[0].device_path = "/dev/ttyS0";
    b->uart_dev[1].rx = 2;
    b->uart_dev[1].tx = 3;
    b->uart_dev[1].device_path = "/dev/ttyS1";

Would that work?

arfoll, 2015-07-22:
yeah technically you can use that Uart however you could also use the FTDI header to provide this? Why not just use that? if going non standard Arduino might as well simplify the crazy muxing and you don't lose the Aio pins.

You'd want something like this because you need to flip the muxes which are 77 & 76. You'd lose the Aio pins obviously.
```
b->uart_dev[1].rx = 2;
b->uart_dev[1].rx.mux_total = 1;
b->uart_dev[1].rx.mux[0].pin = 77;
b->uart_dev[1].rx.mux[0].value = 1;
b->uart_dev[1].tx = 3;
b->uart_dev[1].tx.mux_total = 1;
b->uart_dev[1].tx.mux[0].pin = 76;
b->uart_dev[1].tx.mux[0].value = 1;
b->uart_dev[1].device_path = "/dev/ttyS1";
```

If you test something like this and send it as a PR I'll merge it.

xthunderheartx, 2015-07-22:
The problem with using the FTDI header is that the hardware guy's built our shield (more of a daughter-board than a shield really) based on what the documentation said (or at least implied) and because mechanically using the Arduino header was better apparently.  I don't know that sort of thing so can't comment on that.  Point is changing it causes a re-spin of our hardware which is costly and makes the hardware boys go right round the bend.

We use UART0 to talk to socket modems of various flavors and UART1 to talk to a PIC which interfaces to a number of ASONG 1-wire temp/humidity sensors.  So I gotta figure out a way to get it to work if at all possible.

More importantly since libmraa 0.7.3 will now return a proper context from mraa_uart_init (1), why would you not expose the pins? Isn't that what mraa_uart_init (1) is supposed to do?  I mean why else would I want to call mraa_uart_init for ttys1?  It's already setup, I could just stop the console getty, setup termios and get busy right?  The only reason to initialize it from mraa is so I can talk to it over the Arduino header.

Maybe I'm not thinking about this properly but it doesn't seem logical to me not to have both Galileo UARTs work the same way as far as the mappings go since that is what they are advertised to do.  I'll see if I can get your suggestion above to work.

arfoll, 2015-07-22:
Well the idea is that people doing a mraa_uart_init(1) may want to still use A2/A3 as analog ports. Whilst yes you could use termios we also enable the mraa_uart_write() stuff.

Why not just use mraa_gpio_init_raw() to expose the two muxes and set them manually? That should be easy for you and you won't need to change your hardware.

xthunderheartx, 2015-07-23:
Fair enough.  I see your point. I'll try to get it up manually per your suggestion.

xthunderheartx, 2015-07-24:
Tried your suggestion but no joy.  Is there something I need to do other that setting the values of gpio77 / gpio76 high?  Checked sysfs after my code runs and both pin show up and have a value of '1' but get no data on the header. Thought maybe my test code might be bad (though it works fine with /dev/ttys0) so I just stuck an ftdi2usb dongle on pins 2&3.  Conceivably I should be able to toggle power, set the values of gpio76/77 to '1' and see console output on the header right?

arfoll, 2015-07-27:
So it's a bit more complicated than I thought, what you need is to set up your gpios like this, so 16 must be OUT LOW and 17 OUT HIGH. then 76 & 77 both OUT HIGH. I've verified this works for me on a gen2 running an IOTDK 1.5 image.

```
root@galileo:~# cat /sys/kernel/debug/gpio 
GPIOs 16-31, i2c/0-0025, pcal9555a, can sleep:
 gpio-16  (sysfs               ) out lo
 gpio-17  (sysfs               ) out hi

GPIOs 64-79, pca9685-gpio, can sleep:
 gpio-76  (sysfs               ) out hi
 gpio-77  (sysfs               ) out hi
```

xthunderheartx, 2015-07-27:
Actually Brendan I came to the same conclusion about gpio16/17.  You said you verified it worked for you?  For me ... nope:

```
root@Garviel:~# cat /sys/kernel/debug/gpio
GPIOs 0-1, platform/sch_gpio.2398, sch_gpio_core:

GPIOs 2-7, platform/sch_gpio.2398, sch_gpio_resume:

GPIOs 8-15, intel_qrk_gip_gpio:
 gpio-8   (SPI_CS              ) out hi
 gpio-9   (pcal9555a-exp2-int  ) in  hi

GPIOs 16-31, i2c/0-0025, pcal9555a, can sleep:
 gpio-16  (sysfs               ) out hi
 gpio-17  (sysfs               ) out hi

GPIOs 32-47, i2c/0-0026, pcal9555a, can sleep:

GPIOs 48-63, i2c/0-0027, pcal9555a, can sleep:

GPIOs 64-79, pca9685-gpio, can sleep:
 gpio-76  (sysfs               ) out hi
 gpio-77  (sysfs               ) out hi
```

Nothing on IO3, but outputting "A" on the ftdi header.

?????????????  I don't see how it could **not** work.  You physically verified you can rx/tx over those pins correct?

arfoll, 2015-07-27:
Your 16 is out high not out low. And yes I plugged in an ftdi cable rx/tx/gnd to pin 2/3 and read/wrote.

arfoll, 2015-07-28:
Triple tested and pushed instructions + sample code. My file looks like this:
```
root@galileo:~# cat /sys/kernel/debug/gpio 
GPIOs 0-1, platform/sch_gpio.2398, sch_gpio_core:

GPIOs 2-7, platform/sch_gpio.2398, sch_gpio_resume:

GPIOs 8-15, intel_qrk_gip_gpio:
 gpio-8   (SPI_CS              ) out hi
 gpio-9   (pcal9555a-exp2-int  ) in  hi

GPIOs 16-31, i2c/0-0025, pcal9555a, can sleep:
 gpio-16  (sysfs               ) out lo
 gpio-17  (sysfs               ) out hi

GPIOs 32-47, i2c/0-0026, pcal9555a, can sleep:
 gpio-34  (sysfs               ) out hi
 gpio-47  (sysfs               ) out hi

GPIOs 48-63, i2c/0-0027, pcal9555a, can sleep:
 gpio-63  (sysfs               ) in  hi

GPIOs 64-79, pca9685-gpio, can sleep:
 gpio-76  (sysfs               ) out hi
 gpio-77  (sysfs               ) out hi
```

Instructions are here: https://github.com/intel-iot-devkit/mraa/blob/master/docs/galileorevh.md (formating is completely off look at the raw file, I'll fixup formatting later)

closed 2015-07-28 by arfoll

xthunderheartx, 2015-07-28:
Sheez I clipped the wrong text.  But trust me dood I ain't makin' this stuff up just to waste your time or mine:

https://drive.google.com/file/d/0B6ta-n23GeWQSlhwMERkcTRnV0E/view?usp=sharing

Here's the mapping code.  I stepped though it to make sure there were no failures.

```
	mraa_gpio_context	gpio16;	// IO3 level shifter
	mraa_gpio_context	gpio17;	// IO3 pullup

	// First, set the Level/Pullup mode for IO2/IO3

	gpio16 = mraa_gpio_init_raw (16);
	gpio17 = mraa_gpio_init_raw (17);

	result = mraa_gpio_dir (gpio16, MRAA_GPIO_OUT_HIGH);
	result = mraa_gpio_dir (gpio17, MRAA_GPIO_OUT_HIGH);
```

As you can see in the video, sysfs shows them both high and the data is *still* coming over the ftdi header.  The only thing I see that is truly different than you're example is you're using python and my code is C.  Does python use the underlying Linux serial i/o or is writing to the hardware with calls directly to mraa?  My code actually opens the device, and then calls ```tcsetattr``` and then reads/writes to the open file descriptor.  Oh man that line of reasoning is so flawed it's ridiculous.  I'm grasping @ straws here ...

I guess we just have to give up and run a wire from the ftdi header to our board, though it galls me to do it.  Taking it off the Arduino header is a far more elegant solution and it should work.  At least your example does clear up the mapping mystery and others may benefit from it.

Thanx for your help!

arfoll, 2015-07-28:
This is still wrong, you need this:
```
mraa_gpio_dir (gpio16, MRAA_GPIO_OUT_HIGH);
```
to be:
```
mraa_gpio_dir (gpio16, MRAA_GPIO_OUT_LOW);
```

The serial output will still be on the FTDI header, it'll be on A2/A3 as well. The muxing doesn't actually disable the FTDI header I don't think that's possible. There's no difference using python or C, copy my python code and try that. In my setup I have two FTDI USB->serials one on the FTDI header and one on A2/A3, they both see the same thing.

Don't worry I fully understand your frustration, the muxing can get confusing on this platform.

xthunderheartx, 2015-07-28:
OMG!  ROFL!  I'm sorry Brendan.  Apparently I need a weekend at the beach or something.  Ok 16 low, 17 high.  Got it.

arfoll, 2015-07-29:
No problem, we all need a holiday :). I'm assuming all working now?