eclipse-mraa/mraa #245: Rpi A+ - failed to initialise platform

leonyuhanov, 2015-07-22:
I have a Rpi A+
 node Version v0.11.0
 MRAAV -> v0.7.3

I am initialing SPI:

var m = require('mraa');
console.log("MRAA ->\t"+m.getVersion());

And i get this error:
mraa: FATAL error, failed to initialise platform
MRAA -> v0.7.3

arfoll, 2015-07-22:
That's odd, can you share your kernel log (journalctl or /var/log/messages)? I've honestly never seen an A+.

@michael-ring have you got one of these?

michael-ring, 2015-07-22:
I should have one somewhere, will check in the evening. 

@leonyuhanov Could you also please do:

cat /proc/cpuinfo

and add this output to this issue?

TnX

leonyuhanov, 2015-07-22:
Will do in an hour

leonyuhanov, 2015-07-22:
Ok:
@michael-ring :

pi@RPWOW01 ~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 2.00
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2708
Revision	: 0012
Serial		: 00000000f19a309f

@arfoll re the message file I dont know how to attach it to this message so i have put it on my web site: http://www.elec-tron.org/wp-content/uploads/2015/07/messages.txt

arfoll, 2015-07-22:
@leonyuhanov are you sure your user 'pi' has access to /dev/spidev*? There's less interesting stuff than I was hoping tbh.

FYI I typically use dpaste.com is a good way of sharing logs (though your site is just as good)

leonyuhanov, 2015-07-22:
@arfoll 
>>are you sure your user 'pi' has access to /dev/spidev
how can i check that? the user "pi" is the standard user that comes out of the box with the device..

arfoll, 2015-07-22:
Check with something like `ls -lah /dev/spide*`. Also run python with 'sudo', root will have access for sure.

leonyuhanov, 2015-07-22:
pi@RPWOW01 /dev $ ls -lah /dev/spide*
crw-rw---T 1 root spi 153, 0 Jan  1  1970 /dev/spidev0.0
crw-rw---T 1 root spi 153, 1 Jan  1  1970 /dev/spidev0.1

Im using MRAA with NodeJS. i ran node as sudo:
pi@RPWOW01 ~/rpwowsuit $ sudo ../nodejs/bin/node test.js
starting

mraa: FATAL error, failed to initialise platform
MRAA ->	v0.7.3

same thing

leonyuhanov, 2015-07-22:
FYi the test script suggested on the rpi doco, https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md,  work fine:

pi@RPWOW01 ~/spitest $ ./spidev_test -D /dev/spidev0.0
spi mode: 0
bits per word: 8
max speed: 500000 Hz (500 KHz)

00 00 00 00 00 00 
00 00 00 00 00 00 
00 00 00 00 00 00 
00 00 00 00 00 00 
00 00 00 00 00 00 
00 00 00 00 00 00 
00 00

tingleby, 2015-07-22:
Did you install via npm?

leonyuhanov, 2015-07-22:
@tingleby 
Yes
i downloaded the latest build of node js as a tar and extracted it to /home/pi/nodejs
pi@RPWOW01 ~/nodejs/bin $ ./node --version
v0.11.0
pi@RPWOW01 ~/nodejs/bin $ ./npm --version
1.2.15
I then cd /home/pi
npm install mraa
it does its thing, compiles and and its there.
i copied the module folders into my project directory

leonyuhanov, 2015-07-22:
I also tried node v0.10.x and had the same problem
should i try and compile nodejs v12 from source? and then try it all again?

arfoll, 2015-07-22:
Ah, then that's the issue. ARM platforms arent' supported via NPM - tbh I thouhgt it didn't build flat out so hadn't thought of that... that's one of those long standing things I haven't fixed. Sorry! Please see https://github.com/intel-iot-devkit/mraa/issues/79

leonyuhanov, 2015-07-22:
>>Try building from source with cmake/swig etc that'll work.

any chance you could help me out here I'm struggling with all this linux stuff... i dont know how to do what you suggested..

arfoll, 2015-07-22:
Try follow the building instructions - https://github.com/intel-iot-devkit/mraa/blob/master/docs/building.md

if you get any issues post here.

leonyuhanov, 2015-07-22:
ok thanks ill give it a go

leonyuhanov, 2015-07-22:
ok so followed the build instructions, everything looks liek it worked. I now have:
pi@RPWOW01 ~/mraa/build $ ls -la
total 68
drwxr-xr-x  6 pi pi  4096 Jul 22 11:45 .
drwxr-xr-x 12 pi pi  4096 Jul 22 11:44 ..
-rw-r--r--  1 pi pi  1600 Jul 22 11:45 arch.c
-rw-r--r--  1 pi pi 15952 Jul 22 11:45 CMakeCache.txt
drwxr-xr-x  6 pi pi  4096 Jul 22 11:48 CMakeFiles
-rw-r--r--  1 pi pi  1864 Jul 22 11:45 cmake_install.cmake
-rw-r--r--  1 pi pi   291 Jul 22 11:45 CTestTestfile.cmake
drwxr-xr-x  4 pi pi  4096 Jul 22 11:47 examples
-rw-r--r--  1 pi pi 14710 Jul 22 11:45 Makefile
drwxr-xr-x  5 pi pi  4096 Jul 22 11:46 src
drwxr-xr-x  3 pi pi  4096 Jul 22 11:45 tests

where is the actual "mraa" node_module folder that i include in my project?

leonyuhanov, 2015-07-22:
fyi http://www.elec-tron.org/wp-content/uploads/2015/07/buildlog.txt this is the build log

arfoll, 2015-07-22:
Sadly you need to update swig before you'll build the node.js module.

See the warning in your log:
`WARN - SWIG is 2.0.7. Please upgrade to 3.0.5+ to build nodejs addon`

Either update your image on your pi or build swig from source (it's not particularly big or difficult).

leonyuhanov, 2015-07-22:
This is like an onion of frustration!
Ill try and get Swig 3 working on this bloody thing
thanks for your help

leonyuhanov, 2015-07-23:
@arfoll hey mate
im abandoning using mraa, its far to difficult to install on the rpi. I was hoping to replace the shitty Intel boards i had been using with the rpi and port my code over, but it looks like I'm going to have to do allot of work. This https://www.youtube.com/watch?v=J_Oe3ELh-bI is my current project that uses a Galileo Gen 2 and mraa, if your interested. I only need SPI, and will attempt to find a different nodejs library that will work with the rpi A+ and SPI. I appreciate your time with this query and past ones with the Galileo boards on the Intel Forums.

closed 2015-07-23 by leonyuhanov