eclipse-mraa/mraa #246: Npm fails intel edison

raunaqrox, 2015-07-23:
I did `npm install mraa --verbose`
my node version is v0.10.28
npm version is 1.4.9

```
gyp verb extracted file from tarball deps/v8/src/incremental-marking-inl.h
gyp verb extracted file from tarball deps/v8/src/flags.h
gyp verb extracted file from tarball deps/v8/src/dateparser-inl.h
gyp verb content sha1 node-v0.10.28.tar.gz 944ae69de8106ee2f54607fafc16a25b6950ae80
gyp verb tarball done parsing tarball
gyp verb check download content sha1, need to download `SHASUMS.txt`... 
gyp verb `SHASUMS.txt` url http://nodejs.org/dist/v0.10.28/SHASUMS.txt
gyp http GET http://nodejs.org/dist/v0.10.28/SHASUMS.txt
gyp http 200 http://nodejs.org/dist/v0.10.28/SHASUMS.txt
gyp verb `SHASUMS.txt` data {"node-v0.10.28-darwin-x64.tar.gz":"6eff580cc8460741155d42ef1ef537961194443f","node-v0.10.28-darwin-x86.tar.gz":"d95ed3e8305ba459015549ef8a925ffbb889ef35","node-v0.10.28-linux-x64.tar.gz":"4b9cf9437decea3d9913b694ea2e9b0a06ced2dd","node-v0.10.28-linux-x86.tar.gz":"3daec68e972abc299882b73fa6360387c5145a97","node-v0.10.28-sunos-x64.tar.gz":"17898dcf188d4b672bf5d2ec2f8e7a0abb6ccbcf","node-v0.10.28-sunos-x86.tar.gz":"19075b6ce07fb3a331b2f710ddddd07f86a5a5cd","node-v0.10.28-x86.msi":"0c9ebeecf7cfad99fad1af7e7043fcd6fac8eef9","node-v0.10.28.pkg":"f94bb307f8c4aa408c4702704461533f1d4fa793","node-v0.10.28.tar.gz":"ef08a75f6359a16e672cae684e0804ca7f4554b7","node.exe":"57e565d7a2c10dd9299c15b98ab752e01ab3b4e0","node.exp":"97de4ea47677773ee21c2361b88fa8751f613bb3","node.lib":"20614aa1b6505e3c6921897bab09117b86da9c29","node.pdb":"10bfcfa936d25a12f0778f472c3439c2aaac0ff9","pkgsrc/nodejs-ia32-0.10.28.tgz":"9aa42387fb7d18a74059ffef85f9d8ca316525b6","pkgsrc/nodejs-x64-0.10.28.tgz":"b987cd71d9ca1aeb513343b459a3e5f7b8c4df1d","x64/node-v0.10.28-x64.msi":"6122b245f2237268a3ad10e8b008752888713906","x64/node.exe":"8017b24d8e6955bbfc2dc28f7a57c3d03e159020","x64/node.exp":"8da812d32830442a0aafa3c3be85ca3ee7ce43c4","x64/node.lib":"8961ea28d99d05dd85c3bf1ac4dd88fb4e23be3b","x64/node.pdb":"02d57f356d754da2e5ba8885243ad7a9a67bad38"}
gyp verb download contents shasums {"node-v0.10.28.tar.gz":"944ae69de8106ee2f54607fafc16a25b6950ae80"}
gyp verb validating download shasum for node-v0.10.28.tar.gz (944ae69de8106ee2f54607fafc16a25b6950ae80 == ef08a75f6359a16e672cae684e0804ca7f4554b7)
gyp WARN install got an error, rolling back install
gyp verb command remove [ '0.10.28' ]
gyp verb remove using node-gyp dir: /home/root/.node-gyp
gyp verb remove removing target version: 0.10.28
gyp verb remove removing development files for version: 0.10.28
gyp ERR! configure error 
gyp ERR! stack Error: node-v0.10.28.tar.gz local sha1 944ae69de8106ee2f54607fafc16a25b6950ae80 not match remote ef08a75f6359a16e672cae684e0804ca7f4554b7
gyp ERR! stack     at deref (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:296:20)
gyp ERR! stack     at IncomingMessage.<anonymous> (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/install.js:336:13)
gyp ERR! stack     at IncomingMessage.EventEmitter.emit (events.js:117:20)
gyp ERR! stack     at _stream_readable.js:919:16
gyp ERR! stack     at process._tickCallback (node.js:419:13)
gyp ERR! System Linux 3.10.17-poky-edison+
gyp ERR! command "node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/root/projects/node_modules/mraa
gyp ERR! node -v v0.10.28
gyp ERR! node-gyp -v v0.13.0
gyp ERR! not ok 
npm verb unsafe-perm in lifecycle false
npm info mraa@0.7.3 Failed to exec install script
npm info /home/root/projects/node_modules/mraa unbuild
npm info preuninstall mraa@0.7.3
npm info uninstall mraa@0.7.3
npm verb true,/home/root/projects/node_modules,/home/root/projects/node_modules unbuild mraa@0.7.3
npm info postuninstall mraa@0.7.3
npm ERR! mraa@0.7.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mraa@0.7.3 install script.
npm ERR! This is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 3.10.17-poky-edison+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install" "mraa" "--verbose"
npm ERR! cwd /home/root/projects
npm ERR! node -v v0.10.28
npm ERR! npm -v 1.4.9
npm ERR! code ELIFECYCLE
npm verb exit [ 1, true ]
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/root/projects/npm-debug.log
npm ERR! not ok code 0
```
Tried many times does not seem to work.
Thanks.

arfoll, 2015-07-23:
I've just built it on my end, but this looks like an npm/node.js bug/url issue not mraa because you're getting this:

```
gyp verb check download content sha1, need to download `SHASUMS.txt`... 
gyp verb `SHASUMS.txt` url http://nodejs.org/dist/v0.10.28/SHASUMS.txt
gyp http GET http://nodejs.org/dist/v0.10.28/SHASUMS.txt
gyp http 200 http://nodejs.org/dist/v0.10.28/SHASUMS.txt
gyp verb `SHASUMS.txt` data {"node-v0.10.28-darwin-x64.tar.gz":"6eff580cc8460741155d42ef1ef537961194443f","node-v0.10.28-darwin-x86.tar.gz":"d95ed3e8305ba459015549ef8a925ffbb889ef35","node-v0.10.28-linux-x64.tar.gz":"4b9cf9437decea3d9913b694ea2e9b0a06ced2dd","node-v0.10.28-linux-x86.tar.gz":"3daec68e972abc299882b73fa6360387c5145a97","node-v0.10.28-sunos-x64.tar.gz":"17898dcf188d4b672bf5d2ec2f8e7a0abb6ccbcf","node-v0.10.28-sunos-x86.tar.gz":"19075b6ce07fb3a331b2f710ddddd07f86a5a5cd","node-v0.10.28-x86.msi":"0c9ebeecf7cfad99fad1af7e7043fcd6fac8eef9","node-v0.10.28.pkg":"f94bb307f8c4aa408c4702704461533f1d4fa793","node-v0.10.28.tar.gz":"ef08a75f6359a16e672cae684e0804ca7f4554b7","node.exe":"57e565d7a2c10dd9299c15b98ab752e01ab3b4e0","node.exp":"97de4ea47677773ee21c2361b88fa8751f613bb3","node.lib":"20614aa1b6505e3c6921897bab09117b86da9c29","node.pdb":"10bfcfa936d25a12f0778f472c3439c2aaac0ff9","pkgsrc/nodejs-ia32-0.10.28.tgz":"9aa42387fb7d18a74059ffef85f9d8ca316525b6","pkgsrc/nodejs-x64-0.10.28.tgz":"b987cd71d9ca1aeb513343b459a3e5f7b8c4df1d","x64/node-v0.10.28-x64.msi":"6122b245f2237268a3ad10e8b008752888713906","x64/node.exe":"8017b24d8e6955bbfc2dc28f7a57c3d03e159020","x64/node.exp":"8da812d32830442a0aafa3c3be85ca3ee7ce43c4","x64/node.lib":"8961ea28d99d05dd85c3bf1ac4dd88fb4e23be3b","x64/node.pdb":"02d57f356d754da2e5ba8885243ad7a9a67bad38"}
gyp verb download contents shasums {"node-v0.10.28.tar.gz":"944ae69de8106ee2f54607fafc16a25b6950ae80"}
gyp verb validating download shasum for node-v0.10.28.tar.gz (944ae69de8106ee2f54607fafc16a25b6950ae80 == ef08a75f6359a16e672cae684e0804ca7f4554b7)
gyp WARN install got an error, rolling back install
```

I'd suggest updating to the latest edison image, it's on node.js 0.10.38 & NPM 1.4.28 and that seems to work fine.

raunaqrox, 2015-07-23:
ok thanks, I'll try it and let you know. The image is latest though!

arfoll, 2015-07-23:
Download the image here - http://downloadmirror.intel.com/25028/eng/edison-image-ww25.5-15.zip. I'm 99% sure that has node.js 0.10.38.

alext-mkrs, 2015-09-03:
I've just updated one of my Edisons to ww25.5 image and you can wind it up to 100% - the image has node 0.10.38 :smile:

```
root@edison:~# node --version
v0.10.38
```

alext-mkrs, 2015-09-24:
If this one is a "wontfix" and it's two months old, shouldn't we close it, @arfoll or @tingleby?

alext-mkrs, 2015-10-26:
Closing this one for no activity & solution posted. @raunaqrox give us a shout if you still need help with this one.

closed 2015-10-26 by alext-mkrs

mulderp, 2015-12-09:
I just see similar problems with nvm and install of Node 4.0

```
root@edison:~# nvm install 4.0
######################################################################## 100.0%
-sh: sha1sum: command not found
tar: unrecognized option '--strip-components'
BusyBox v1.22.1 (2015-06-19 07:36:41 CEST) multi-call binary.

Usage: tar -[cxtZzJjahmvO] [-X FILE] [-T FILE] [-f TARFILE] [-C DIR] [FILE]...

Binary download failed, trying source.
Additional options while compiling:
```

alext-mkrs, 2015-12-10:
This actually doesn't look like a related issue :smileys: Error messages reveal that the command you run tries to execute sha1sum command, which apparently isn't installed on your board. Then it also tries to use tar with parameters that Busybox doesn't support. So you'd need to install both utilities first.