eclipse-mraa/mraa #247: mraa causing nodejs script to crash, cannot kill proccess

leonyuhanov, 2015-07-29:
I posted this on the Intel support forum as well:
I have an Edison with a Mini Breakout Board. Running the latest IOT image with whatever version of mraa it comes with.
At completely random intervals, initiating or using the SPI module in my code causes my app to become dormant. I can not kill it via the terminal and have to reboot the board.

The process looks like this after its dead in TOP:

365   351 root     D    37760   4%   0% node main.js 

and this pops up in the console after a minute or so:

root@edison:/node_app_slot# node main.js  
[  240.618710] INFO: task node:365 blocked for more than 120 seconds.  
[  240.618800] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.  
This repeats, i can no longer run any node apps, and i have to reboot the device!

This occurs ONLY when i implement the MRAA module.
All i have to do is is include these 2 lines:

var m = require('mraa');  
x = new m.Spi(0);  

And 90% of the time the script just dies as above

leonyuhanov, 2015-07-31:
FYI I am on the latest version of MRAA already. I ran the following code to test:
var m = require('mraa');  
x = new m.Spi(0);  
console.log("LIBMRAA Versoin details\n--->\t\t"+m.getVersion());  
The resulting out put was:
root@edison:/node_app_slot# node main.js  
LIBMRAA Versoin details  
--->            v0.7.3

leonyuhanov, 2015-08-01:
Update, i have a feeling this issue is specifically with the Edison and not with the MRAA module. Have tested another SPI NodeJS driver and am getting the exact same failure.

closed 2015-08-01 by leonyuhanov

arfoll, 2015-08-01:
Yes that looks kernel related, check the output of `dmesg` and report on the edison forums. Cheers

kersten, 2015-09-21:
could you paste the link here? I have the exact same problem and need a fix

leonyuhanov, 2015-09-21:
@kersten hey mate, Intel has confirmed this is a problem with the latest release of their Edison firmware. They have purposely kneecapped the spi driver to not work over 20khz. Have a look at this https://communities.intel.com/thread/78149?sr=inbox, its the forum post where we confirmed with Intel that this problem exists. It is not an issue with the mraa library

leonyuhanov, 2015-09-21:
And here is where i opened a support request with intel
https://communities.intel.com/thread/77705?sr=inbox

leonyuhanov, 2015-09-21:
This is what Intel support told me after 2 weeks of waiting on their support:
"Intel Edison uses the SPI driver available with the Yocto embedded Linux. Like most embedded devices the SPI driver is emulating SPI interface protocol use bitbang technique therefore the speed on the SPI is limited. Only device with hardware SPI interface can achieve high speed SPI. By increasing the SPI speed the SPI driver provided by Yocto Linux get overwhelmed with the data and crashes, causing the kernel panic."
At this stage, i disconnected my Edison board and put it in the Rubish bin where it belongs

leonyuhanov, 2015-09-21:
If you really want to get it working, downgrade your Edison board to the previous firmware. You will need to update mraa and will be stuck with old software. This is the link https://software.intel.com/en-us/iot/hardware/edison/downloads you need the v2 release, anything newer is broken

kersten, 2015-09-21:
@leonyuhanov thanks for your detailed answer. This is very sad. You know some other microcontroller boards that will not have this problems? I have some beaglebones and raspberry pis at home. Should they work?

> At this stage, i disconnected my Edison board and put it in the Rubish bin where it belongs

Maybe I should have done this too. At this time I did not get any LED strip working like expected and this is the only thing I want to do, light up multiple strips. For now I am using a bbb with the LEDscape code.

leonyuhanov, 2015-09-21:
Yep im using a rpi A+ succesfully with mraa to drive apa102 strips.
If u want something smaller, try the NodeMCU modules or teensys
Leon

kersten, 2015-09-21:
Thanks. I first tried to get ws2812b strips to work over firmata / arduino, but this was not really fast enough. Working with one strip was OK, but talking to my ONLY six strips was horrible. Now I got one apa102 5m test strip and thought I can drive this one with the edison.

Did I understand the forum post correctly that the edison just emulates SPI and does not have a chip for it on board?

leonyuhanov, 2015-09-21:
that is what Intel is saying yea. Wether ita true or not we will never know. Fyi i was able to drive 2000 x apa102 leds with the edison on v2 firmware.. it can definitely do it, but some idiot at intel decided that they kill spi, i3c etc.. look on their forums,people are abandoding ship.
Leon

kersten, 2015-09-21:
Ok, then I will try that. Is there anything I have to watch out for? I read that you nearly broke your device.

leonyuhanov, 2015-09-21:
If u use an rpia+ follow theae to the T http://www.elec-tron.org/?p=996
If u want to use the ediaon with v2 firmware, just reflash it . I didnt encounter any huge issues from memory
Leon

ivictbor, 2016-02-29:
Still have the same issue on Edison, thanks for explanations. The edison firmware is very bugged now, CS0 on SPI also not works at all, MRAA hangs, I am waiting for next bug...

leonyuhanov, 2016-02-29:
@ivictbor throw out your Edison, its garbage.

alext-mkrs, 2016-03-01:
@ivictbor, there were several spi fixes in Edison's kernel repo, see "01org" on GitHub. If you're up to that, you can build your image with those patches by following Edison's bsp build guide.

arfoll, 2016-03-01:
FYI the response from Intel support is incorrect, edison does not use an upstream driver from 'yocto linux' or anything like that. It also does not bitbang a SPI bus, it's using another piece of hardware (i believe it's some kind of UART HW) to emulate a SPI bus. This driver however does have some limitations but has been improved notably as was mentioned by @alext-mkrs.

@ivictbor you haven't mentioned what you're doing exactly so we can't really help you much more, why don't you open a new issue.

@leonyuhanov let's keep it constructive :smiley:

leonyuhanov, 2016-03-01:
@arfoll I wasted allot of time and money on Intel boards and got sick and tired of their s###t, Im just saying it how it is mate. For 1/4 the price the Rpi will do more than any of their boards AND mraa works as intended

ivictbor, 2016-03-06:
@arfoll Ok, thanks, I performed some experiments and wrote detailed issue here: https://github.com/intel-iot-devkit/mraa/issues/456

ivictbor, 2016-03-06:
@alext-mkrs Can you please tell me more details about 01org, is I need to use this repo https://github.com/01org/edison-linux ? I tried to build yocto using this guide http://www.intel.com/content/dam/support/us/en/documents/edison/sb/edisonbsp_ug_331188007.pdf and next archive http://downloadmirror.intel.com/25028/eng/edison-src-ww25.5-15.tgz . Building for core-image-minimal runs ok, but there are no bitbake edison-image target as in guide. Do I need another guide or another source archive ? And is there some instructions how to build yocto with patches from 01org?