eclipse-mraa/mraa #251: i2c writeReg() failure on MinnowboardMax

tripzero, 2015-08-06:
Here's my output from testing:

I'm testing with the max and the pca9685 PWM controller from adafruit.  writeReg() returns INVALID_HANDLE (5).

```
>>> import mraa
>>> i = mraa.I2c(0)
>>> i.address(0x40)
0
>>> i.readReg(0x00)
0L
>>> i.writeReg(0x00, 0x31)
5
>>> quit()
```

i2cset, however seems to work:

```
root@intel-corei7-64:~# i2cset 7 0x40 0x00 0x31
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will write to device file /dev/i2c-7, chip address 0x40, data address
0x00, data 0x31, mode byte.
Continue? [Y/n] y
root@intel-corei7-64:~#
```

arfoll, 2015-08-07:
Can you use i2c in raw mode to check its not an 0 != 7 issue?

Otherwise not sure, can you paste the output of journalctl &| syslog?

tripzero, 2015-08-07:
Tried with i = mraa.I2c(0, raw=True), same result.  Below is the journalctl output:

```
Aug 07 21:09:47 intel-corei7-64 libmraa[504]: libmraa version v0.7.3 initialised by user 'root' with EUID 0
...
Aug 07 21:18:49 intel-corei7-64 libmraa[504]: i2c: Failed to write
Aug 07 21:18:58 intel-corei7-64 libmraa[504]: i2c: Failed to write
```

arfoll, 2015-08-07:
You'd want:

mraa.I2c(7, True)

tripzero, 2015-08-10:
```root@intel-corei7-64:~# python
Python 2.7.9 (default, Aug  4 2015, 16:36:12) 
[GCC 4.9.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
>>> i = mraa.I2c(7, True)
>>> i.writeReg(0x00, 0x31)
5
```

arfoll, 2015-08-11:
What mraa version are you running exactly? Any patches? Can you try with master and see if you can reproduce? I'd also like to see syslog output.

tripzero, 2015-08-11:
Testing master without any patches I get:

```
import mraa
i = mraa.I2c(0) 
i.address(0x40)
5
i = mraa.I2c(7, True)
i.address(0x40)
0
```

journalctl sees: 

Aug 11 17:14:32 intel-corei7-64 libmraa[495]: libmraa version v0.7.3 initialised by user 'root' with EUID 0
Aug 11 17:14:41 intel-corei7-64 libmraa[495]: i2c: Failed to open requested i2c port /dev/i2c-1264
Aug 11 17:14:41 intel-corei7-64 libmraa[495]: i2c: Failed to get I2C_FUNC map from device
Aug 11 17:14:48 intel-corei7-64 libmraa[495]: i2c: Failed to set slave address 64

closed 2015-08-20 by arfoll

SovatnaPhon, 2018-11-16:
I installed mraa, while it is problem for me with Invalid I2c bus. Can you help me?

commit 9b9a5ab1 (2015-08-20): intel_minnow_max.c: Fix default i2c bus mechanism

Fixes #251. default_i2c_bus was set to 7 before setting actual parameters for
it leading to obvious failure as 0 != 7