eclipse-mraa/mraa #252: Compiling mraa from source on RPi A+ for use as a NodeJS module

leonyuhanov, 2015-08-11:
I previously gave up on using mraa on my rpia+ but had some time and wanted to give it another go.

Here is my pre installation instructions
http://www.elec-tron.org/?p=996

Here is my make log
http://www.elec-tron.org/wp-content/uploads/2015/08/makelog.txt
Seems to have completed correctly
Where do i find the actual mraa NODEJS module directory? or is there another step i need to do?

arfoll, 2015-08-11:
Hey, thanks for giving us another chance :) you're nearly there! The issue with your instructions (it's hard to know for sure because you didn't include the log of ```cmake ..```) is that when you configure swig you need to do ```./configure --prefix=/usr```. Then rerun cmake/make and it will build a .node file.

leonyuhanov, 2015-08-11:
hey mate...so i found
~/mraa/build/src/javascript/mraa.node
and copied that into my 
~/project/node_modules/mraa/build/Release

and what do you know, it bloody well works!!!
Cheers

closed 2015-08-11 by leonyuhanov