eclipse-mraa/mraa #254: nodejs environment, mraa/uart functions not able to read data that does not end with '\n'

wubi4292, 2015-08-14:
If incoming data ends with '\n', SerialPort.dataAvailable() returns true.
If incoming data does not end with '\n', SerialPort.dataAvailable() returns false.

 var SerialPort=new mraa.Uart(0);
 var devPath=SerialPort.getDevicePath();
 var SerialPort=new mraa.Uart("devPath"); 
 console.log(SerialPort.dataAvailable(1000));

arfoll, 2015-08-14:
Are you sure you have baud rate setup correctly? I just tested this and it works as expected:

```
In [6]: x = m.Uart("/dev/ttyS0")
In [7]: x.setBaudRate(115200)
In [18]: x.read(10)
Out[18]: bytearray(b'ds')
In [19]: x.dataAvailable()
Out[19]: True
In [21]: x.read(10)
Out[21]: bytearray(b'ds\r')
```

I just plugged a FTDI 6 pin UART to my PC and used minicom to send 'ds' without carriage return and then sent one with carriage return, see the difference. What do you have on the other end of the serial line?

wubi4292, 2015-08-14:
Thank you! After I changed JS code(line 3, a different name), it works now. Moverover, it could not get data length of received. Not sure if you have comment on it?

var SerialPort=new mraa.Uart(0);
var devPath=SerialPort.getDevicePath();
var sp=new mraa.Uart(devPath);

arfoll, 2015-08-14:
The length of the received data? It's just a normal JS type, grab it's length however you want. It should already have the correct size.

closed 2015-08-14 by arfoll