eclipse-mraa/mraa #258: mraa pin no.9 is invalid GPIO pin?

coloz, 2015-08-22:
i seen the breakout board's mraa number,i want to use no.9
but error said, 'Invalid GPIO pin specified'

alext-mkrs, 2015-08-22:
Are you talking about mini-breakout or an Arduino breakout board here?

A reproducer program listing woud also help a lot in troubleshooting this.

coloz, 2015-08-24:
import mraa
x = mraa.Gpio(9)

it is 'Invalid GPIO pin specified'

i try to use C/C++, it's same error

arfoll, 2015-08-24:
Can you tell us which board, what OS and mraa version (mraa.getVersion() will tell you) and maybe the output of syslog? There's some doc on doing so here: https://github.com/intel-iot-devkit/mraa/blob/master/docs/debugging.md

Without any of this there's not much anyone can do.

alext-mkrs, 2015-09-03:
@coloz is this still actual? If yes, please provide the information @arfoll and I have requested above - that will help us to troubleshoot it. Other wise, please let us know and we'll close this one.

alext-mkrs, 2015-09-20:
@arfoll looks like this one is no longer active, propose closing it.

coloz, 2015-10-06:
@arfoll  breakout board & Release 2.1 Yocto* complete image,thanks a lot

alext-mkrs, 2015-10-06:
@coloz you still haven't provided any useful debug information. "breakout board" is too generic, which one do you use, can you provide e.g. a link to the web shop, if you don't know what the name is? On both Arduino break out board and mini-breakout board gpio pin number 9 is valid, so you shouldn't be getting such an error there, so I suspect you have something else.

Image version is also useless, because it's the library version that matters.

Please see the document @arfoll referred to: https://github.com/intel-iot-devkit/mraa/blob/master/docs/debugging.md and execute commands listed there (to get library version and gather logs), then show us the output.

alext-mkrs, 2015-10-26:
Ok, it's been long enough with no activity - I'm closing this one. @coloz, feel free to add a comment if you still need help with this one.

closed 2015-10-26 by alext-mkrs