eclipse-mraa/mraa #259: [API] I2C Read Function Implementation (Question)

Justin-Mai, 2015-08-22:
Hi everyone,

I'm trying to interface with a HoneyWell HIH6130 Humidity / Temperature sensor, and I was wondering if I could get some clarification on how the i2c functions... function in mraa.

I know this isn;t the greatest place to ask, but I figured I'd try anywhere I could.

So, what I need to know is: what's happening on the data line when I call the mraa_i2c_read function?

Say I give the i2c context and init and i2c address just like in the example file, and I've successfully connected to my device. No I send the command and specify 4 bytes of data to read into a "rx_tx_buf" buffer.

My sensor datasheet says it wants the address of the device, followed by a read signal 91 for read, 0 for write). At which point the sensor sends an "acknowledged" bit (ACK) and 8 bits of data. THEN, the sensors wants an "acknowledged" bit to be sent from the master (Edison) to the slave (sensor) in order for the sensor to send the next byte of data. 

This is the case between all 4 bytes of data; it wants the master to acknowledge the data it's reading. 

Does the mraa_i2c_read() function respond with acknowledgement bits after reading each byte from the slave?

Also, since there is a bit of data (acknowledged bit) the slave is sending before the first actual byte of sensor data, will the last bit of the data byte be "chopped off" by the function? Basically, if there's actually 9 bits of response from the slave, will the mraa function read only the first 8 bits and end up "shifting" the rest of the data over 1 bit-length?

Here's a picture to illustrate the call-response format the sensor requires: http://i.imgur.com/n6MJAiu.png

Thank you all so much in advance. We're trying to get an Edison to wirelessly stream data from a satellite, and it's my first time working with the sensor, an Edison, and programming C++ - it's quite the learning curve!

arfoll, 2015-08-24:
mraa/linux will handle sending the ACK and read/write signal in i2c addr for you so as long as your sensor follows the i2c standard there should be nothing for you to worry about there.

What you want is:

alext-mkrs, 2015-10-25:
@Justin-Mai, have you been able to communicate with the sensor, do you need any further help with this i2c communication?

alext-mkrs, 2015-10-29:
Closing due to no activity and answer provided. Feel free to let us know if you need further help with this.

closed 2015-10-29 by alext-mkrs