eclipse-mraa/mraa #261: Xadow GPIO mapping not clear, perhaps not working?

deadprogram, 2015-08-25:
When trying to use the Xadow breakout board with an Edison, I am unable to access some of the pins that would appear should be addressable.

I am trying to use the buzzer and/or vibration motor, that require P1 & P2, or P2 & P4:
http://www.seeedstudio.com/wiki/Category:Xadow

From this page of pin mappings here:
http://www.seeedstudio.com/wiki/Xadow_IO_pin_mapping

I would expect to want to use either: 

```
var bz1 = new m.Gpio(15);
var bz2 = new m.Gpio(5);
```

or else: 

```
var bz1 = new m.Gpio(15);
var bz2 = new m.Gpio(16);
```

However both of the above result in errors such as this error:
```
var bz2 = new m.Gpio(5);
          ^
Error: Illegal arguments for construction of _exports_Gpio
    at Object.<anonymous> (/home/root/blink/buzzer.js:4:11)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:935:3
```

How should I access each of the required GPIO pins when using MRAA on the Intel Edison with the Xadow? Or have I uncovered a bug?

Thank you!

arfoll, 2015-08-25:
Xadow breakout is treated like a miniboard so so gpio 5 is invalid, that's normal. However 15 works here for me on Xadow. 16 is also invalid as http://iotdk.intel.com/docs/master/mraa/edison.html.

However where I can't help you is that I have no idea how the Xadow board is wired up and which GPIO is which.

deadprogram, 2015-08-25:
Here is a link to the schematic for the Xadow Edison breakout board, in case that can prove to be helpful in determining the mappings:

http://www.seeedstudio.com/wiki/images/3/3b/Xadow-Edison_sch_v1.0.pdf

deadprogram, 2015-09-13:
Hi, @arfoll any further into on this?

arfoll, 2015-09-14:
No really, all I can tell you is that the 'xadow breakout' has i2c-6 (so '0' in mraa) broken out on the header if you plug it on the header with the 'power' side (on the other header it doesn't seem to work). I'll try get in contact with Seed to ge them to tell us because randomly probing is just a waste of time...

deadprogram, 2015-09-14:
I was able to easily get the i2c working. The GPIO, not so much. Thanks for looking into it.

omkarkhair, 2016-05-08:
I did a looping around all Gpio pins this weekend, to find `new mraa.Gpio(6)` working on the Xadow breakout board SCL pin on J4.