eclipse-mraa/mraa #264: Memory Leak in nodejs module on RPIA+

leonyuhanov, 2015-08-29:
Team I'm running into an old issue where im getting a memory leak by using the mraa.write(buffer).
My code sends about 4000 bytes in 1 frame at 4mhz at teh moment. and after about 2 minutes runtime the program goes form 40mb in top to 80mb... I ran into this problem on the Galileo Gen2 board but it was fixed previously...I checked the source code on my rp and it matches the fix...but still having this issue

leonyuhanov, 2015-08-30:
Update...
Not 100% sure this is a problem with MRAA... May be a problem with iether NodeJS or the SPI driver on the RpiA+..
I ran 2 bits of code:
1: Send out 128Bytes of data in a tight while loop:
while(true)
{
 spiDevice.write(dataBuffer);
}
RAM usage climbs until I kill the script or until the Device crashes. 
2: send out 128Bytes of data in a tight loop but delay each write with a console.log or a delay function
while(true)
{
 spiDevice.write(dataBuffer);
console.log("delaying with concolelog");
}
Ram usage is steady...
3: send out 1000 blocks of 128 bytes of data in a tight loop

var counter=0
while(true)
{
 spiDevice.write(dataBuffer);
counter++;
 if(counter>1000)
 {
 counter=0; 
 console.log("delaying with text");
 }
}
RAM usage climbs whilst data is transmitted, then after the 1st pause, drops and stays steady...

arfoll, 2015-08-30:
Run your code with `node --expose-gc test.js` and then call the garbage collector explicitly at certain moments `global.gc();`. The proper solution would be to tune node.js's garbage collector to be more aggresive by default on small platform. Hopefully this solves your problem!

leonyuhanov, 2015-08-30:
Tried that before posting my original issue. I think the problem is with Node as ive tested a different node spi module and got far worse memory leaks than mraa

closed 2015-08-30 by leonyuhanov

arfoll, 2015-08-30:
Yeah sadly large data transfers from userspace do end up with a bunch of memcpy's so if node doesn't clean itself up on time it gets out of hands very quickly. If the memory comes back after a while then usually that smells of garbage collection issue :/

You could try add swap to your platform at least that would give you a bit more buffer room.

leonyuhanov, 2015-08-30:
All good, ive put in small pauses inbetween large transmitions to resolve this issue. Have had my app running for 4 hours with no rise in ram usage. Cheers