eclipse-mraa/mraa #269: i think there is a discrepance between PWM functions name and what they do

cosscat, 2015-09-02:
Did this on Edison, but i guess is valid for other platforms
in python:
```
>>> import mraa as m
>>> p=m.Pwm(3)
>>> p.enable(True)
>>> p.period(0.2)
0
p.pulsewidth_ms(5)
0
```

What this does is blink an LED with a frequency of 5 Hz.
The pulsewidth_ms() makes the LED dimmer or brighter depening on the no. of ms
It is behaving like the 5Hz frequency is a "carrier" frequency for a much higher frequency, that is the pulses of light themselves are the result of an other inner pwm. Hope i explained this well.

alext-mkrs, 2015-10-02:
Hey @cosscat 
What would you expect to see here and where it's different to what you actually see?

`period()` sets a period for PWM, that is 200 ms in your example. `pulsewidth_ms()` sets a pulse width, in milliseconds, 5 ms in your example. Taken together that means that PWM would take plase each 200 ms and out of those the pin will be in HIGH state for 5 ms. And that results in that blinking with 5 Hz frequency, as a direct result of the period set.

To me this does look like a perfectly proper behavior, thus function names also look ok. Please elaborate on "expected vs. actual", that would probably help me to understand why you see this as a problem/discrepancy.

alext-mkrs, 2015-10-21:
@cosscat, do you think that's still actual or my explanation resolves it for you?

alext-mkrs, 2015-10-29:
Closing due to no activity and answer provided. Feel free to comment if you need further help.

closed 2015-10-29 by alext-mkrs