eclipse-mraa/mraa #272: npm install error on Intel Edison

Vayvala, 2015-09-05:
Intel Edison freshly updated to release 2.1 Yocto

```
root@va:~/js# uname -a
Linux va 3.10.17-poky-edison+ #1 SMP PREEMPT Fri Jun 19 12:06:40 CEST 2015 i686 GNU/Linux
root@va:~/js# npm install mraa
..........
make: Leaving directory '/home/root/js/node_modules/mraa/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:12)
gyp ERR! System Linux 3.10.17-poky-edison+
gyp ERR! command "node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/root/js/node_modules/mraa
gyp ERR! node -v v0.10.38
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok

npm ERR! mraa@0.7.4 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the mraa@0.7.4 install script.
npm ERR! This is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 3.10.17-poky-edison+
npm ERR! command "node" "/usr/bin/npm" "install" "mraa"
npm ERR! cwd /home/root/js
npm ERR! node -v v0.10.38
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0
```

arfoll, 2015-09-07:
NPM module is broken currently on node.js <0.12.0. I'll release a new version asap to fix this.

arfoll, 2015-09-07:
This'll need a new release which I'll do later today to make sure it's in npm.

closed 2015-09-07 by arfoll

arfoll, 2015-09-07:
Should be fixed with 0.7.5 that's uploaded. Tested on node 0.10.x both on 64 and 32bit.

commit dc6a5859 (2015-09-07): binding.gyp.cmake: Fix npm builds for ia32 node.js 0.10.x

Node.js 0.10.x seems to struggle with the 'or' directive. Fixes #272