eclipse-mraa/mraa #281: gpio isr not working if pin was used previously

cosscat, 2015-09-17:
Tried on Galileo v2 with mraa ver. 'v0.7.5-55-g5c7bab1'
```
>>> io=m.Gpio(10)
>>> io.dir(m.DIR_OUT)
0
>>> io.dir(m.DIR_IN)
0
>>> io.read()
0
>>> io.read()
1
>>> io.dir(m.DIR_OUT)
0
>>> io.write(0)
0
>>> io.write(1)
0
>>> def func():
...    print "isr function accessed "
...
>>> io.isr(m.EDGE_FALLING, func, 6)
0
>>>
```

trying here to touch pin 10 with 5V and nothing happens.
then:
```
>>> io.isrExit()
0
>>> io.isr(m.EDGE_RISING, func, 6)
0
>>>
```
Again, nothing.

The same happens if gpio is instantiated with a new pin(and not using it in advance for something):
```
>>> io=m.Gpio(9)
>>>
>>> io.isr(m.EDGE_FALLING, func, 6)
0
>>> io.isrExit()
0
>>> io.isr(m.EDGE_RISING, func, 6)
0
>>>
```
 At this point isr is not functional.

tingleby, 2015-09-17:
The first part, did you set the GPIO back to Input mode?

cosscat, 2015-09-17:
Hi, 
Yes i did. Forgot to add that part though. Here is a new try:
```
>>> io=m.Gpio(9)
>>> def func():
...   print "hello"
...
>>> io.dir(m.DIR_IN)
0
>>> io.isr(m.EDGE_RISING, func, 6)
0
```

and again, nothing happens

cosscat, 2015-09-17:
... same on Minnow-Max

cosscat, 2015-09-18:
same for Galileo v2 and Edison.

alext-mkrs, 2015-10-02:
Hi @cosscat 

With Python you need to put a `sleep()` at the end of the ISR-related script for it to work properly, like in the [example](https://github.com/intel-iot-devkit/mraa/blob/master/examples/python/hello_isr.py). See [this commit](https://github.com/intel-iot-devkit/mraa/commit/dcc4efbaf86a8908a84d9363968fe23365b2691a) (especially the docs piece) for details, but in general the reason is that in Python the destructor kicks in almost at the same time as the ISR started and it just ends the whole process (including the ISR thread) before it's established.

alext-mkrs, 2015-10-25:
@cosscat, do you think we can close this one?

alext-mkrs, 2015-10-29:
Closing due to no activity and solution provided.

closed 2015-10-29 by alext-mkrs