eclipse-mraa/mraa #283: unwanted behavior of PWM on Minnow Max

cosscat, 2015-09-17:
Hello,

Noticed an unwanted behavior of PWM on Minnow Max: 'v0.7.5-55-g5c7bab1'
Try the following:
```
>>> p=m.Pwm(22)
>>> p.enable(True)
0
>>> p.config_percent(10, 0.5)
0
>>> p.config_percent(10, 0.8)
0
>>> p.config_percent(10, 0.2)
0
>>> p.config_percent(10, 0.8)
0
>>> p.config_percent(10, 0.9)
0
>>> p.config_percent(10, 0.8)
0
>>> p.config_percent(10, 0.9)
0
>>> p.config_percent(10, 0.8)
0
>>> p.config_percent(10, 0.9)
0
>>> p.config_percent(10, 0.7)
0
>>> p.config_percent(10, 0.5)
0
>>> p.config_percent(10, 0.9)
```
where the first parameter is the  period in ms and the second the percentage of "on" state.
You will notice (for an LED) that when doing the p.config_percent(10, 0.9) it will light less than on 
p.config_percent(10, 0.5)

cosscat, 2015-09-17:
same happens for:
>>> p.enable(True)
0
>>> p.period(0.1)
>>> p.pulsewidth(0.05)
0
>>> p.pulsewidth(0.09)

The same behavior is valid for the second PWM pin: 24
It is something related to 90% duty cycle, no matter how one sets it.

alext-mkrs, 2015-10-26:
Unfortunately I don't have a MinnowBoard check this out, @cosscat are you able to reproduce this on either generation of Galileos or an Edison (all of which I have)?

cosscat, 2015-12-02:
Sorry for the late reply,

Things are fine with Galileos ... will update shortly for Edison

cosscat, 2015-12-03:
Edison is fine too