eclipse-mraa/mraa #284: Minnow Max unwanted behavior for pulsewidth_ms()

cosscat, 2015-09-17:
Hello,

Tested on: 'v0.7.5-55-g5c7bab1'
The behavior of pulsewidth_ms() chaotic. An LED used for it on pin 22 would not follow the duty cycle settings:
```
>>> p = m.Pwm(22)
>>> p.period(0.1) # this is 100ms
0
>>> p.pulsewidth_ms(90) # LED not lighting
0
>>> p.pulsewidth_ms(90) # LED lights
0
>>> p.pulsewidth_ms(80) # LED is not lighting
0
>>> p.pulsewidth_ms(70) # LED lights
0
>>> p.pulsewidth_ms(60)
0
>>> p.pulsewidth_ms(50)
0
>>> p.pulsewidth_ms(90)

```

cosscat, 2015-09-17:
happens for p.write(percentage value) too:
eg. p.write(0.2)

cosscat, 2015-09-17:
the percentage is set correctly:
>>> p.read()
0.89999997615814209

alext-mkrs, 2015-10-26:
Same as with #283, if you could reproduce this on Galileo or Edison I'd be glad to look into this, otherwise we'd need to wait until @arfoll has time to check it out. Though I'm not sure if you have a MinnowBoard either, Brendan?

cosscat, 2015-12-02:
Works fine for Galileo, will check for Edison shortly

cosscat, 2015-12-03:
Edison is fine too

alext-mkrs, 2015-12-03:
Got it, thanks @cosscat. Then unfortunately we'll need to wait for someone with Minnowboard to try this out as I don't have one as I mentioned.