eclipse-mraa/mraa #289: UART flush() not working as expected on Edison

cosscat, 2015-09-18:
Hello,
mraa ver.: 'v0.7.5-55-g5c7bab1'
for uart module:
write something in Putty and then:
```
>>> u.dataAvailable()
True
>>> u.flush() 
0
>>> u.dataAvailable()
True
```
I didn't wrote something else in between those two u.dataAvailable() and the flush should have flushed the buffer.

arfoll, 2015-09-18:
we already went through this on IM but tcdrain() which is what is called by flush() simply blocks until everything is written as writes are async. It does not discard buffers as tcflush() would do. Tbh I'm not sure I like the behaviour but it's whats done in all the other serial libs I've seen :)

closed 2015-09-18 by arfoll

cosscat, 2015-09-18:
yes, you're right. i forgot this one. if it's ok for you i can consider this an exception for Edison