eclipse-mraa/mraa #290: Python's API's I2c.address() does not produce expected results

vivitter, 2015-09-21:
Hi, I'm a beginner of Edison.
Now I'm trying to get data from Bosch BME280, combined humidity and pressure sensor.
https://ae-bst.resource.bosch.com/media/products/dokumente/bme280/BST-BME280_DS001-10.pdf

I've found a sample code for arduino.
https://github.com/SWITCHSCIENCE/BME280/tree/master/Arduino/BME280_I2C

It worked successfully.
Next, I would like to replace this arduino code to python with mraa.
I wrote the code below.
(It is alternative execution of the line 78-84 in the arduino sketch.)

```python
import mraa
m = mraa.I2c(6)
m.address(0x76)
m.readRegByte(0x88)
```

It returns the result below.

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/mraa.py", line 893, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, I2c, name)
  File "/usr/lib/python2.7/site-packages/mraa.py", line 78, in _swig_getattr
    return _swig_getattr_nondynamic(self, class_type, name, 0)
  File "/usr/lib/python2.7/site-packages/mraa.py", line 73, in _swig_getattr_nondynamic
    return object.__getattr__(self, name)
AttributeError: type object 'object' has no attribute '__getattr__'
```

After that, I tested this code in python interpreter step by step.
The result shows that "m.address(0x76)" returns 0.

Do you have any suggestions to debug this?
Thank you for your attention.

### Appendix

```
# i2cdetect -y -r 6
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- 76 --
```

```
# i2cdump -y 6 0x76
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
80: 94 8f 89 1d 97 00 33 06 96 6d 1b 67 32 00 53 8d    ?????.3??m?g2.S?
90: 21 d6 d0 0b ae 11 99 00 f9 ff ac 26 0a d8 bd 10    !??????.?.?&????
a0: 00 4b 8d 00 00 00 00 00 00 00 00 00 33 00 00 c0    .K?.........3..?
b0: 00 54 00 00 00 00 60 02 00 01 ff ff 1f 71 03 00    .T....`?.?..?q?.
c0: 00 00 27 ff 00 00 00 00 01 00 00 00 00 00 00 00    ..'.....?.......
d0: 60 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    `...............
e0: 00 62 01 00 15 02 00 1e 76 41 ff ff ff ff ff ff    .b?.??.?vA......
f0: ff 00 01 00 27 a0 00 60 55 00 83 df 00 7b e1 80    ..?.'?.`U.??.{??
```

arfoll, 2015-09-21:
So m.address() simply sets the address of the i2c context for future operations. It doesn't actually do anything.

What version of mraa are you using? (check with m.getVersion()). You may want to update to 0.7.5 but this should just work...

arfoll, 2015-09-21:
Oh wait readRegByte doesn't exist. Use 'readReg()'

vivitter, 2015-09-21:
Oops! It was all my fault!
It worked correctly.

Thank you @arfoll !!

closed 2015-09-21 by vivitter

Tamiroquai, 2016-12-21:
Hi Man,
Can you share the complete python lib, if you have it?

arfoll, 2016-12-21:
If you want to use the bme280 I'd recommend looking at the relevant upm
library see
http://iotdk.intel.com/docs/master/upm/classupm_1_1_b_m_e280.html

There a python binding for it as for every upm lib.