eclipse-mraa/mraa #301: iio: Allow attribute/raw channel reading

arfoll, 2015-10-02:


alext-mkrs, 2015-10-10:
I think this one would require some sort of DAC device to be available, to be able to test it, right?

arfoll, 2015-10-10:
For the chanel writing yes, but I really meant this function https://github.com/intel-iot-devkit/mraa/blob/iio-api/src/iio/iio.c#L181 which would implement the possibility of writing to attributes.

alext-mkrs, 2015-10-11:
Yep, I see, just double-checking if I'm able to take this on - I unfortunately don't have anything DAC.

I'll check if I can implement this without that.

laykuanloon, 2015-10-16:
i implement one for event using string, i do the same for mraa_iio_write and tested write into scale is working
mraa_iio_write(iio_device0, "in_accel_scale", "0.038326");

Should I check in?

alext-mkrs, 2015-10-16:
I think @arfoll meant actual integers/floats, not a string. If you could adjust it like that, then feel free to send it in, no sense for me to do the same.

Or you can create a temporary branch with whatever you have as-is an I'll take from there in a day or two.

arfoll, 2015-10-19:
@laykuanloon Can you make a new PR for that function? That looks useful, we can make it more generic afterwards, I can take it in the iio-api branch and we'll improve it later.

arfoll, 2017-02-07:
Closed as this is done

closed 2017-02-07 by arfoll