eclipse-mraa/mraa #303: Cross compile Libmraa for intel Edison from Linux x86_64

badr-badri, 2015-10-02:
Hello,
Could you please provide me the command line that allows me to compile the libmraa from my linux (Debian jessie x86_64) for intel Edison.
Thanks in advance.

arfoll, 2015-10-02:
This should have everything you need:

https://github.com/intel-iot-devkit/mraa/blob/master/docs/building.md

badr-badri, 2015-10-02:
I'm sorry but it's not very clear for me, what are the options to pass cmake to indicate the architecture i686 of intel edison ?

badr-badri, 2015-10-02:
When I do this : $cmake -DBUILDARCH=i586 then make then make install, the result still x86_64 library "libmraa.so.0.8.0: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV)"

arfoll, 2015-10-02:
If you are cross compiling are using the or cross cmake toolchain file? You
can't just set target arch and have it work on edison....

Brendan

badr-badri, 2015-10-02:
I don't use it cause i don't know how can use it. That's why i'm asking for help. Could you please give me the instructions that I have to follow ?

tingleby, 2015-10-03:
The following should what you need to cross compile at least the C/C++ using cmake and the Yocto toolchain.

OK so first things first you will need a toolchain that matches what is running on the Edison. Intel provides one from.
https://software.intel.com/en-us/iot/hardware/edison/downloads
Download the "SDK - Cross Compile Tools" and as you're using Debian x86_64 you want "Linux* 64-bit"
```shell
wget http://downloadmirror.intel.com/25028/eng/edison-sdk-linux64-ww25.5-15.zip
unzip edison-sdk-linux64-ww25.5-15.zip  
./poky-edison-glibc-x86_64-edison-image-core2-32-toolchain-1.7.2.sh
```
You will then be prompted where you want to install the SDK to, the default is `/opt/poky-edison/1.7.2` and is what I will use for these instructions.
After completion you should have everything you need to cross compile for the Edison Yocto  distro.

The next step "sources" a script provided by the Yocto SDK which will add the compilers,sysroots to your running environment variables. This is tested under BASH and ZSH other shells may work but no promises :)
```shell
source /opt/poky-edison/1.7.2/environment-setup-core2-32-poky-linux
```

So, now its time to compile mraa!
```shell
git clone https://github.com/intel-iot-devkit/mraa.git
cd mraa
mkdir build
cd build
cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchains/oe-sdk_cross.cmake ../
make
```
What should have happened now is that you used the included OE toolchain file in mraa to use the enviroment and sdk you set up in the previous steps. Alll going well you have now compiled mraa for edison

badr-badri, 2015-10-03:
It works :) Thank so much for your help, I really appreciate it :)

badr-badri, 2015-10-08:
Hi guys,
I have another question, which variables should I change so I can get a deb package (with checkinstall) so I can install it in ubilinux with dpk -i ?
Thanks in advance.

tingleby, 2015-10-08:
So, the instruction I provided are targeting Yocto images provided by Intel. There will probably be a different way for targeting ubilinux, which I don't know exactly what is.

badr-badri, 2015-10-08:
Thanks for your response, I appreciate :)

badr-badri, 2015-10-08:
Do you know the instructions to cross compile for debian i386 ? This should work according to ubilinux forum

tingleby, 2015-10-08:
So, I may be wrong and these suggestions are *untested*!.

You will probally need to following packages
`gcc-multilib libc6-i386 lib6-dev-i386`

* Use your toolchain avaible on your host, gcc-multilib
* `rm cmake/modules/CPackDeb.cmake`
* Call cmake like this `cmake-DCMAKE_CXX_FLAGS=-m32 -DCMAKE_C_FLAGS=-m32 -DBUILDSWIG=off ../`

This might be partly the way todo it.

tingleby, 2015-10-08:
The rm command is to remove the modifications that are in mraa so it packages for IPK over DEBs. Removing use the standard debian behaviour, at least it should.

badr-badri, 2015-10-08:
it doesn't work actually but never mind, i'll find a way 
Thanks a lot for taking time to respond :)

closed 2015-10-26 by arfoll