eclipse-mraa/mraa #305: Analog reading at 8khz

sriranjanr, 2015-10-04:
Hello,

I want to analog read A0 pin at 8khz. How shall I get a fast read out? I am dealing with a project where I need to sample the microphone on the ADC at 8khz.I am using the intel arduino edison breakout board. Someone please tell me how to.

arfoll, 2015-10-04:
You'd be better off asking on the intel maker forums this is not the right place to ask, you need to sample your ADC directly from IIO. We're working on mechanisms to do this from mraa but it's not yet available. Try libiio.

closed 2015-10-04 by arfoll

sriranjanr, 2015-10-04:
The ADC used on edison arduino breakout board is ADS7951.So what kind of driver it might need.Is it possible to use MRAA to improve the ADC frequency?

arfoll, 2015-10-04:
You can't take a measurement at 8khz from Linux userspace accurately with the current mraa support, you would need the upcoming trigger buffer IIO support which the ADS7951 edison driver doesn't actually support and even then I'm not sure you'd be able to do that well.

Since you're trying to sample a microphone my advice would be to get a cheap USB soundcard instead, that will do the job much better & more easily.

sriranjanr, 2015-10-05:
Could you shed more  light on how to use IIO?

alext-mkrs, 2015-10-05:
As @arfoll mentioned, IIO is only a theoretical solution, because (a) real performance is not known and (b) Edison's ADC driver doesn't support the necessary mode anyway.

So using a usb sound card is really the best solution here.

sriranjanr, 2015-10-05:
I am thinking of using an external ADC with software SPI to sample and capture the mic data. SPIdev transfer calls are having 50us and 45us delays going in and out of the calls.

alext-mkrs, 2015-10-05:
That's your call of course  but imho that will cost you more than a usb sound card, with the same result