eclipse-mraa/mraa #315: MRAA build foler for UPM

xueliu, 2015-10-09:
As building instruction said, the building of UPM needs following information.

PKG_CONFIG_PATH=$PKG_CONFIG_PATH:.../mraa/build/lib/pkgconfig
CPLUS_INCLUDE_PATH=$CPLUS_INCLUDE_PATH:.../mraa/build/include
LIBRARY_PATH=$LIBRARY_PATH:.../mraa/build/lib

But after I compile the mraa, I can not find lib/, include/ and lib/ folder under build/.

Where or how can I find such folders ? Thx

alext-mkrs, 2015-10-09:
What exactly do you mean by "I compile the mraa", which command do you run?
To have this directory structure in place you need to run "make install" and set `CMAKE_INSTALL_PREFIX` and/or `DESTDIR` variables (otherwise the default location will be `/usr` - this equals to `../mraa/build` part in the UPM instruction).

You can see the install logic and locations in `src/cmake_install.cmake` in CMake's build directory, which you chosse when running CMake.

xueliu, 2015-10-10:
yes, you are right. The problem is solved.

closed 2015-10-10 by xueliu

alext-mkrs, 2015-10-10:
Glad to help :smiley: