eclipse-mraa/mraa #318: Implemented check to detect Arduino or Mini Breakout Board is not sufficient / fails

GitAnonymous, 2015-10-13:
Using mraa with my custom board fails since for some reasons my custom board is being detected as an Arduino Breakout Board instead of a Mini Breakout Board.

The actual board check is performed in intel_edison_fab_c.c:

------------------------------------------------------------------------------
mraa_board_t*  
mraa_intel_edison_fab_c()  
{  
    mraa_board_t* b = (mraa_board_t*) calloc(1, sizeof(mraa_board_t));  
    if (b == NULL) {  
        return NULL;  
    }  
  
  
    b->platform_name = PLATFORM_NAME;  
    // This seciton will also check if the arduino board is there  
    tristate = mraa_gpio_init_raw(214);  
    if (tristate == NULL) {  
        syslog(LOG_INFO, "edison: Failed to initialise Arduino board TriState,\  
                assuming Intel Edison Miniboard\n");  
        if (mraa_intel_edison_miniboard(b) != MRAA_SUCCESS) {  
            goto error;  
        }  
        return b;  
    }  
[...]
------------------------------------------------------------------------------

As you can see the whole decision is based on initializing GPIO 214. Maybe  some further checks would provide a better stability.

If mraa recognizes the wrong board the application will crash before reaching "main" with "Segmentation fault".

This topic has been discussed here: https://communities.intel.com/message/339192

tingleby, 2015-10-13:
@arfoll Perhaps we could look for the ADC? Or scan the i2c-bus with the TI muxes on it?

wwhuang, 2016-02-01:
Hi, has there been any progress on this? We've run into the same problem on a custom board we have, and we've resorted to commenting out the check and forcing mraa to assume that we are not an arduino.

alext-mkrs, 2016-02-01:
I don't think there was any work done on this one yet. I have it on my list for the next weekend though.

closed 2016-02-12 by arfoll

uchobby, 2016-02-18:
I'd suggest that we have a way to override the detection, a configuration option, an environment variable, or at the very least an API call into MRAA to fix it when it's wrong.

arfoll, 2016-02-18:
I'm not sure how it would help though. If mraa detects it wrong then likely your platform is in a state where we'll just bork it even further by doing what we expect to work and won't. I suspect we'll get something like this eventually for some subplatforms that can't be autodetected and to provide custom platform files as is being worked described in json but until then I don't really see the use case.

wwhuang, 2016-02-18:
Custom platforms is exactly the use case I have in mind. My lab has its own Edison based hardware platform which mraa detects as an Arduino. A JSON file or mraa api call would be fantastic.

commit b33bd5cc (2016-02-12): intel_edison_fab_c.c: Enhanced detection of Arduino expansion board

Closes #318.