eclipse-mraa/mraa #323: Any bluetooth examples?

rclai, 2015-10-19:
Was wondering if there was a way to programmatically work with the bluetooth module?

alext-mkrs, 2015-10-19:
You probably mean Edison here, right?

Generally speaking, mraa is not targeting this, because (IMHO, @arfoll would be authoritative here) there's not much can be done to make it simpler (given all those profiles you have there), which is mraa's main goal. To use bluetooth, you could use standard linux BT programming approaches, involving BlueZ most probably.

Though while typing this I started to think that actually be an interesting idea to try and implement something like this, maybe in some limited fashion, but still that could make BT more accessible :smiley:

rclai, 2015-10-19:
Yes, the Edison. I actually found another solution by using [this solution](https://github.com/bbx10/node-nRF8001), which depends on Noble. So I'm closing now.

closed 2015-10-19 by rclai