eclipse-mraa/mraa #329: build libmraa for Pi and BB

jiangzidong, 2015-10-22:
If I want to build libmraa for Raspberry Pi and BB, what should I do?
just in ubuntu and `cmake --DBUILDARCH=arm`?

arfoll, 2015-10-22:
You only need to set the architecture like that if you are cross compiling
or forcing it. If you are compiling on the target then it will detect the
arch.

tingleby, 2015-10-22:
@arfoll no if during the cmake setup it sees a compatible arch, will it not set to that. So if you use gcc-arm it will automatically select arm platforms? The forcing is mainly a testing feature.

mulderp, 2015-10-25:
Also, it might be to see the commands to fetch dependencies before running a build. E.g. for Yocto distributions, there are these commands:

    echo "src mraa-upm http://iotdk.intel.com/repos/1.1/intelgalactic" > /etc/opkg/mraa-upm.conf
    opkg update
    opkg install libmraa0

But on RPi or BB one would probably use some kind of "apt-get" commands?

leonyuhanov, 2015-10-26:
I wrote up a step by step guide here
http://www.elec-tron.org/?p=996

arfoll, 2015-10-26:
@mulderp our instructions are only valid for iot devkit images galileo/edison/minnowboard max, probabyl should make this clearer

jiangzidong, 2015-11-18:
Thanks, I will close the issue for now.

closed 2015-11-18 by jiangzidong