eclipse-mraa/mraa #330: SPI: JavaScript, SPI.write function prototype

xbolshe, 2015-10-22:
JavaScript, Node v4.1.1,  mraa/examples/javascript/spi.js

SPI.write prototype:
uint8_t * write (uint8_t *txBuf, int length)

NodeJS code:
x = new m.Spi(0)
buf2 = x.write(buf)
console.log("Sent: " + buf.toString('hex') + ". Received: " + buf2.toString('hex'))

Question: what is an output of the function: received data length or received data buffer ?

arfoll, 2015-10-22:
received data buffer - is it not working as you'd expect?

xbolshe, 2015-10-22:
Yes, an output is a noise, but it shall be as an input.
I have checked kernel driver and libmraa - no problem.
The problem is in mraajs.i
[image]

xbolshe, 2015-10-22:
BTW, mraajs.i has the following code:

namespace mraa {
class Spi;
%typemap(out) uint8_t*
{
%#if SWIG_V8_VERSION > 0x040000
   v8::MaybeLocal<v8::Object> objret = node::Buffer::New(v8::Isolate::GetCurrent(), (char*) $1, arg3);
   free($1);
   if(!objret.ToLocal(&$result)) {
      SWIG_exception_fail(SWIG_ERROR, "Spi buffer failed");
      SWIGV8_RETURN(SWIGV8_UNDEFINED());
   }

Please pay your attention for %typemap(out) uint8_t*
It is not a buffer!

xbolshe, 2015-10-22:
https://github.com/intel-iot-devkit/mraa/pull/332

xbolshe, 2015-10-22:
With changes I have:

[image]

xbolshe, 2015-10-22:
And need to add free(result); to free uint8_t* recv = malloc(sizeof(uint8_t) * length); from mraa_spi_write_buf

commit 63ff9016 (2015-10-26): JavaScript SPI.write output correction #2

Closes #330.
closed 2015-10-26 by arfoll