eclipse-mraa/mraa #331: Error when installing on Edison from master branch

hansmosh, 2015-10-22:
I have a fresh install of Ubilinux and I followed these install steps:
https://learn.sparkfun.com/tutorials/installing-libmraa-on-ubilinux-for-edison
(with one addition of adding a python 3 flag `cmake .. -DBUILDSWIGNODE=OFF -DBUILDPYTHON3=ON`)

Install goes fine but I can't `import mraa`.

It works when I install from the tag v0.7.3. I chose that one sort of randomly because I knew this worked a few months ago.

```Python 3.4.2 (default, Oct  8 2014, 13:14:40)
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.4/site-packages/mraa.py", line 32, in <module>
    _mraa = swig_import_helper()
  File "/usr/local/lib/python3.4/site-packages/mraa.py", line 28, in swig_import_helper
    _mod = imp.load_module('_mraa', fp, pathname, description)
  File "/usr/lib/python3.4/imp.py", line 243, in load_module
    return load_dynamic(name, filename, file)
ImportError: /usr/local/lib/python3.4/site-packages/_mraa.so: undefined symbol: PyString_AsString```

alext-mkrs, 2015-10-23:
That looks strange, because there's no such function in Python 3 indeed, it's from version 2.

It looks like if you've built the module against 2 and are trying to run it on 3. My first guess would be that you have a -dev package for 2 instead of 3 (because that would explain it and the instructions you refer to have a reference to namely that version).

What do the following commands return on your Edison:

1. dpkg -l |grep python

hansmosh, 2015-10-23:
1.
```nio@Edison-2:~$ dpkg -l |grep python
ii  dh-python                  1.20141111-2                  all          Debian helper tools for packaging Python libraries and applications
ii  libpython-dev:i386         2.7.9-1                       i386         header files and a static library for Python (default)
ii  libpython-stdlib:i386      2.7.9-1                       i386         interactive high-level object-oriented language (default python version)
ii  libpython2.7:i386          2.7.9-2                       i386         Shared Python runtime library (version 2.7)
ii  libpython2.7-dev:i386      2.7.9-2                       i386         Header files and a static library for Python (v2.7)
ii  libpython2.7-minimal:i386  2.7.9-2                       i386         Minimal subset of the Python language (version 2.7)
ii  libpython2.7-stdlib:i386   2.7.9-2                       i386         Interactive high-level object-oriented language (standard library, version 2.7)
ii  libpython3-dev:i386        3.4.2-2                       i386         header files and a static library for Python (default)
ii  libpython3-stdlib:i386     3.4.2-2                       i386         interactive high-level object-oriented language (default python3 version)
ii  libpython3.4:i386          3.4.2-1                       i386         Shared Python runtime library (version 3.4)
ii  libpython3.4-dev:i386      3.4.2-1                       i386         Header files and a static library for Python (v3.4)
ii  libpython3.4-minimal:i386  3.4.2-1                       i386         Minimal subset of the Python language (version 3.4)
ii  libpython3.4-stdlib:i386   3.4.2-1                       i386         Interactive high-level object-oriented language (standard library, version 3.4)
ii  python                     2.7.9-1                       i386         interactive high-level object-oriented language (default version)
ii  python-apt                 0.8.8.2                       i386         Python interface to libapt-pkg
ii  python-apt-common          0.8.8.2                       all          Python interface to libapt-pkg (locales)
ii  python-cffi                0.8.6-1                       i386         Foreign Function Interface for Python calling C code
ii  python-chardet             2.3.0-1                       all          universal character encoding detector for Python2
ii  python-colorama            0.3.2-1                       all          Cross-platform colored terminal text in Python - Python 2.x
ii  python-cryptography        0.6.1-1                       i386         Python library exposing cryptographic recipes and primitives (Python 2)
ii  python-dev                 2.7.9-1                       i386         header files and a static library for Python (default)
ii  python-distlib             0.1.9-1                       all          low-level components of python distutils2/packaging
ii  python-html5lib            0.999-3                       all          HTML parser/tokenizer based on the WHATWG HTML5 specification (Python 2)
ii  python-minimal             2.7.9-1                       i386         minimal subset of the Python language (default version)
ii  python-ndg-httpsclient     0.3.2-1                       all          enhanced HTTPS support for httplib and urllib2 using PyOpenSSL
ii  python-openssl             0.14-1                        all          Python 2 wrapper around the OpenSSL library
ii  python-pip                 1.5.6-5                       all          alternative Python package installer
ii  python-pkg-resources       5.5.1-1                       all          Package Discovery and Resource Access using pkg_resources
ii  python-ply                 3.4-5                         all          Lex and Yacc implementation for Python2
ii  python-pyasn1              0.1.7-1                       all          ASN.1 library for Python (Python 2 module)
ii  python-pycparser           2.10+dfsg-3                   all          C parser in Python
ii  python-requests            2.4.3-6                       all          elegant and simple HTTP library for Python2, built for human beings
ii  python-setuptools          5.5.1-1                       all          Python Distutils Enhancements
ii  python-six                 1.8.0-1                       all          Python 2 and 3 compatibility library (Python 2 interface)
ii  python-smbus               3.1.1-1                       i386         Python bindings for Linux SMBus access through i2c-dev
ii  python-support             1.0.15                        all          automated rebuilding support for Python modules
ii  python-urllib3             1.9.1-3                       all          HTTP library with thread-safe connection pooling for Python
ii  python-wheel               0.24.0-1                      all          built-package format for Python
ii  python2.7                  2.7.9-2                       i386         Interactive high-level object-oriented language (version 2.7)
ii  python2.7-dev              2.7.9-2                       i386         Header files and a static library for Python (v2.7)
ii  python2.7-minimal          2.7.9-2                       i386         Minimal subset of the Python language (version 2.7)
ii  python3                    3.4.2-2                       i386         interactive high-level object-oriented language (default python3 version)
[... cut ...]
	libgcc_s.so.1 => /lib/i386-linux-gnu/libgcc_s.so.1 (0xb710b000)
	libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xb6f9a000)
	librt.so.1 => /lib/i386-linux-gnu/librt.so.1 (0xb6f91000)
	libexpat.so.1 => /lib/i386-linux-gnu/libexpat.so.1 (0xb6f69000)
	libz.so.1 => /lib/i386-linux-gnu/libz.so.1 (0xb6f4f000)
	libpthread.so.0 => /lib/i386-linux-gnu/libpthread.so.0 (0xb6f34000)
	libdl.so.2 => /lib/i386-linux-gnu/libdl.so.2 (0xb6f2f000)
	libutil.so.1 => /lib/i386-linux-gnu/libutil.so.1 (0xb6f2b000)
	/lib/ld-linux.so.2 (0xb76f9000)```

arfoll, 2015-10-25:
You have two versions of python3 it might have got messed up, can you put your cmake output on dpaste?

alext-mkrs, 2015-10-27:
@hansmosh could you please try with the latest code from master? @arfoll just brought in a potential fix and it would be nice to check if it fixes it for you.

hansmosh, 2015-10-27:
Just installed from the master branch and it's looking good. Thanks! I'm now seeing another issue with the gpio interrupt detection that I will be creating another issue for.

alext-mkrs, 2015-10-28:
Thanks for confirming, sure, we'll take a look a5 the isr one.

closed 2015-10-28 by alext-mkrs