eclipse-mraa/mraa #333: Cannot build NodeJS module in latest Raspberry Pi A+ (Raspbian 4.1)

leonyuhanov, 2015-10-23:
When trying to build mraa from source i run in this error when it gets to compiling the nodejs module:

http://www.elec-tron.org/wp-content/uploads/2015/10/log.txt

These are my installation procedures from scratch -> http://www.elec-tron.org/?p=996 

I have been able to get MRAA working on the previous Raspbian 3.18 kernel but on Raspbian 4.1 i run into the above.... Any suggestions?

alext-mkrs, 2015-10-23:
Looks like a v8 version incompatibility to me. Which node.js and v8 versions do you have on Raspbian 4.1 and on an older one that worked?

leonyuhanov, 2015-10-24:
Yes correct, ill downgrade. Thanks

closed 2015-10-24 by leonyuhanov

leonyuhanov, 2015-10-26:
I have left step by step build instructions on my blog here:
http://www.elec-tron.org/?p=996