eclipse-mraa/mraa #335: Installation fails on Edison with Node.js v4.2.1

michael-p, 2015-10-23:
Dear all,

I'm trying to install mraa on an Intel Edison (running Yocto) using `npm install mraa` but the installation fails (`"... Failed at the mraa@0.8.0 install script 'node-gyp rebuild'. ..."`). I'm not running the preinstalled node.js version (which is an old 0.10 I think) but instead installed node v4.2.1 from source. npm version is 2.14.8. Installing other packages works fine.

Is mraa not (yet) compatible with the most recent node version? If so, are there plans to update it?

Thanks,
Michael

alext-mkrs, 2015-10-23:
Can you please post a full install process output? I wonder if this is the same as #333.

michael-p, 2015-10-24:
Here is the full install output: https://gist.github.com/michael-p/b03bc7f06ad006ff0bdb

I also just rolled back to the old Node version (v0.10.38), installing and using mraa with that version works fine.

xbolshe, 2015-10-24:
mraa 0.8.0 is compatible with node v4.x (Galileo Gen1):

[image]

What SWIG, gcc, g++, cmake, make versions do you use?

arfoll, 2015-10-25:
Need more info from your buildoutput, latest should work with node 4.x as @xbolshe points out

michael-p, 2015-10-25:
Hi guys, thanks a lot for the fast help - really happy to hear that it does work on node 4 ! :)

Here are the version numbers:
```
root@Edison:~# node --version
v4.2.1
root@Edison:~# cat /sys/devices/virtual/dmi/id/board_name
BODEGA BAY
root@Edison:~# node -p "process.versions"
{ http_parser: '2.5.0',
  node: '4.2.1',
  v8: '4.5.103.35',
  uv: '1.7.5',
  zlib: '1.2.8',
  ares: '1.10.1-DEV',
  modules: '46',
  openssl: '1.0.2d' }
root@Edison:~# uname -a
Linux Edison 3.10.17-poky-edison+ #1 SMP PREEMPT Fri Jun 19 12:06:40 CEST 2015 i686 GNU/Linux
root@Edison:~# make --version
GNU Make 4.0
Built for i586-poky-linux-gnu
Copyright (C) 1988-2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
root@Edison:~# python --version
Python 2.7.3
root@Edison:~# g++ --version
g++ (GCC) 4.9.1
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

root@Edison:~# gcc --version
gcc (GCC) 4.9.1
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

root@Edison:~# swig -version

SWIG Version 3.0.5

Compiled with i586-poky-linux-g++  -m32 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mstackrealign -fno-omit-frame-pointer --sysroot=/data/jenkins_worker/workspace/edison-weekly/out/linux64/build/tmp/sysroots/edison [i586-poky-linux-gnu]

Configured options: +pcre

Please see http://www.swig.org for reporting bugs and further information
```

In addition to the output of `npm install mraa` I already posted above here are the contents of npm-debug.log after the installation attempt: https://gist.github.com/michael-p/8a1f41e723ab75f564a8

If you need further information or have ideas what else I could test please let me know!

xbolshe, 2015-10-25:
As I see: "0.8.0":"2015-09-22T16:08:06.701Z" and "modified":"2015-09-22T16:08:06.701Z"
But node 4.x patch was in October. Seems need to update a package in https://registry.npmjs.org/mraa

xbolshe, 2015-10-25:
May you try to execute and post an output?

npm install node-gyp
node-gyp rebuild

michael-p, 2015-10-25:
Ok so I first tried to compile mraa from source (the v0.8.0 tag) but make fails with the following output:
```
...
Scanning dependencies of target mraajs
[ 49%] Building CXX object src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o
In file included from /usr/local/include/node/node.h:42:0,
                 from /home/root/mraa-0.8.0/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:156:
/usr/local/include/node/v8.h:336:1: error: expected unqualified-id before 'using'
 using Handle = Local<T>;
 ^
/usr/local/include/node/v8.h: In constructor 'v8::MaybeLocal<T>::MaybeLocal()':
/usr/local/include/node/v8.h:353:33: error: 'nullptr' was not declared in this scope
   V8_INLINE MaybeLocal() : val_(nullptr) {}
                                 ^
/usr/local/include/node/v8.h: In member function 'bool v8::MaybeLocal<T>::IsEmpty() const':
/usr/local/include/node/v8.h:360:51: error: 'nullptr' was not declared in this scope
   V8_INLINE bool IsEmpty() const { return val_ == nullptr; }
                                                   ^
/usr/local/include/node/v8.h: In member function 'bool v8::MaybeLocal<T>::ToLocal(v8::Local<S>*) const':
/usr/local/include/node/v8.h:364:29: error: 'nullptr' was not declared in this scope
     out->val_ = IsEmpty() ? nullptr : this->val_;
                             ^
/usr/local/include/node/v8.h: In member function 'bool v8::WeakCallbackInfo<T>::IsFirstPass() const':
/usr/local/include/node/v8.h:430:50: error: 'nullptr' was not declared in this scope
   bool IsFirstPass() const { return callback_ != nullptr; }
                                                  ^
/usr/local/include/node/v8.h: At global scope:
/usr/local/include/node/v8.h:469:1: error: expected unqualified-id before 'using'
 using PhantomCallbackData = WeakCallbackInfo<T>;
 ^
/usr/local/include/node/v8.h:472:1: warning: scoped enums only available with -std=c++11 or -std=gnu++11
 enum class WeakCallbackType { kParameter, kInternalFields };
 ^
/usr/local/include/node/v8.h:637:43: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   PersistentBase(PersistentBase& other) = delete;  // NOLINT
                                           ^
/usr/local/include/node/v8.h:638:37: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   void operator=(PersistentBase&) = delete;
                                     ^
/usr/local/include/node/v8.h:814:26: error: expected ',' or '...' before '&&' token
   V8_INLINE Global(Global&& other) : PersistentBase<T>(other.val_) {
                          ^
/usr/local/include/node/v8.h:814:34: error: invalid constructor; you probably meant 'v8::Global<T> (const v8::Global<T>&)'
   V8_INLINE Global(Global&& other) : PersistentBase<T>(other.val_) {
                                  ^
/usr/local/include/node/v8.h:822:40: error: expected ',' or '...' before '&&' token
   V8_INLINE Global& operator=(Global<S>&& rhs) {
                                        ^
/usr/local/include/node/v8.h:844:21: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Global(Global&) = delete;
                     ^
/usr/local/include/node/v8.h:845:29: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   void operator=(Global&) = delete;
                             ^
/usr/local/include/node/v8.h: In constructor 'v8::Global<T>::Global()':
/usr/local/include/node/v8.h:790:42: error: 'nullptr' was not declared in this scope
   V8_INLINE Global() : PersistentBase<T>(nullptr) {}
                                          ^
/usr/local/include/node/v8.h: In member function 'v8::Global<T>& v8::Global<T>::operator=(v8::Global<S>)':
/usr/local/include/node/v8.h:824:18: error: 'rhs' was not declared in this scope
     if (this != &rhs) {
                  ^
/usr/local/include/node/v8.h:827:18: error: 'nullptr' was not declared in this scope
       rhs.val_ = nullptr;
                  ^
/usr/local/include/node/v8.h: In member function 'v8::Global<T> v8::Global<T>::Pass()':
/usr/local/include/node/v8.h:834:44: error: expected '>' before '&&' token
   Global Pass() { return static_cast<Global&&>(*this); }
                                            ^
/usr/local/include/node/v8.h:834:44: error: expected '(' before '&&' token
[... 150 lines cut ...]
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/home/root/mraa-0.8.0/build/src/javascript/build'
make: *** No rule to make target 'Release/obj.target/mraa/src/x86/intel_minnow_byt_compatible.o', needed by 'Release/obj.target/mraa.node'.  Stop.
make: Leaving directory '/home/root/mraa-0.8.0/build/src/javascript/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 3.10.17-poky-edison+
gyp ERR! command "/usr/local/bin/node" "/usr/local/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/root/mraa-0.8.0/build/src/javascript
gyp ERR! node -v v4.2.1
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok 
```

I also see that 3059c06fa05c4a5536ec82d00bd991891956704d from October 3rd mentions node 4 support but release 0.8.0 is from September 22nd. Maybe node 4 support is just not yet on npm?
However, I also tried the above procedure on the master branch and node-gyp rebuild fails there as well (with the same error message).

arfoll, 2015-10-26:
Yes correct, you need master, npm has the latest release version which doesn't support it yet.

closed 2015-10-26 by arfoll

kyrylkov, 2015-10-31:
@arfoll When are you planning to do a release supporting Node.js 4.x.x & 5.x.x ?

tingleby, 2015-10-31:
Ill give what we have on master a go and then tag 0.8.1

kyrylkov, 2015-10-31:
Great!