eclipse-mraa/mraa #336: mraa not properly installing on Minnow Board Max. Running  Debian 8|  Not working in python|

v1v3kdalal, 2015-10-24:
After following the Basic build steps section. I try to use mraa in python. (Tested in 2.7 and 3.4) In 3.4 it does import mraa but as soon as you call a function it gives errors. In 2.7 it won't even import.
The build examples written in cpp do work however.

Any help would be appreciated, i need to use mraa in python.
I have all the python dev's installed.
Thank you. 

Here is the CMakeErrorLog

Determining if the pthread_create exist failed with the following output:
Change Dir: /home/johnson/mraa/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec2530568501/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec2530568501.dir/build.make CMakeFiles/cmTryCompileExec2530568501.dir/build
make[1]: Entering directory '/home/johnson/mraa/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/johnson/mraa/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec2530568501.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec2530568501.dir/CheckSymbolExists.c.o   -c /home/johnson/mraa/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTryCompileExec2530568501
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec2530568501.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTryCompileExec2530568501.dir/CheckSymbolExists.c.o  -o cmTryCompileExec2530568501 -rdynamic 
CMakeFiles/cmTryCompileExec2530568501.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTryCompileExec2530568501.dir/build.make:88: recipe for target 'cmTryCompileExec2530568501' failed
make[1]: Leaving directory '/home/johnson/mraa/build/CMakeFiles/CMakeTmp'
make[1]: *** [cmTryCompileExec2530568501] Error 1
Makefile:118: recipe for target 'cmTryCompileExec2530568501/fast' failed
make: *** [cmTryCompileExec2530568501/fast] Error 2

File /home/johnson/mraa/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/johnson/mraa/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTryCompileExec3021793244/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec3021793244.dir/build.make CMakeFiles/cmTryCompileExec3021793244.dir/build
make[1]: Entering directory '/home/johnson/mraa/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/johnson/mraa/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec3021793244.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTryCompileExec3021793244.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.0/Modules/CheckFunctionExists.c
Linking C executable cmTryCompileExec3021793244
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec3021793244.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTryCompileExec3021793244.dir/CheckFunctionExists.c.o  -o cmTryCompileExec3021793244 -rdynamic -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTryCompileExec3021793244.dir/build.make:88: recipe for target 'cmTryCompileExec3021793244' failed
make[1]: Leaving directory '/home/johnson/mraa/build/CMakeFiles/CMakeTmp'
Makefile:118: recipe for target 'cmTryCompileExec3021793244/fast' failed
make[1]: *** [cmTryCompileExec3021793244] Error 1
make: *** [cmTryCompileExec3021793244/fast] Error 2

arfoll, 2015-10-25:
Try use a pastebin service like dpaste intead of this github messes up all the formatting.

Our cmake config can get confused between python versions, tbh by default it should force python2 usage unless you set -DBUILDPYTHON3=ON. Wipe the build/ dir and try building again with that option and paste your error in python3.

alext-mkrs, 2015-10-25:
Yeah, the output is quite inconvenient to read. What I could derive from that is thet it can't find `pthread_create()` function.

@v1v3kdalal, please post the output of this command: `dpkg -l |egrep 'libc|pthread'`. If it's longer than a couple of lines, please use some service like dpaste.

v1v3kdalal, 2015-10-26:
@alext-mkrs We entered your command here is the output:
http://dpaste.com/37J5BCS 
@arfoll When we built the  cmake files i entered the -DBUILDPYTHON3=ON when i ran make.
However we will see what @alext-mkrs suggests then if needed we will clean the build and reinstall again making sure we use that build command when we build.

v1v3kdalal, 2015-10-26:
@arfoll We wiped our build directory. Then typed make -DBUILDPYTHON3=ON .. then make.
Everything seemed to install correctly.

Then we tried running a MRAA test code in Python 3.4 and this is the error message.
http://dpaste.com/3JM4CC0

alext-mkrs, 2015-10-27:
Ok, so the build problem is gone, now you're trying to run a python script and it sort of fails. Please post your full script, that checksMRAA.py one.

arfoll, 2015-10-27:
@v1v3kdalal I pushed a fix to python3 builds just now, actually we did have a build issue where @alext-mkrs added some useful debug info but makes a call to a function that is python2 only :/. Try update master and build again with commit 083606790061d4c8c70c66007439410ce30de063.

But considering that you are building my guess is you have another issue yet. Please paste the full output of cmake like this: http://dpaste.com/2NCJ5F0

alext-mkrs, 2015-10-27:
That's an interesting one, I actually had no problems compiling that for python 3 :-) And in this one it's a pthread_create(), which was a culprit.

AsString was in the error log for #331.

v1v3kdalal, 2015-10-27:
@alext-mkrs Here is the checkMRAA.py that i run. http://dpaste.com/233Y3P8 
It's error message is: http://dpaste.com/3JM4CC0

Yeah the import works for Python 3 interpreter but then it won't import for the python2.7 interpreter.

In python3 it imports as soon as i call print (mraa.getVersion()) in the checkMRAA.py script it throws the error. Found in http://dpaste.com/3JM4CC0

@arfoll Here is the output of cmake: http://dpaste.com/102T7QC 

Thank you for help.

alext-mkrs, 2015-10-28:
This looks like  a basic import problem, if that's the case it shouldn't be related to the way the module was built, only the installation paths might matter.

Could you please try two things:

1. run the same commands you have in the script using the interactive python shell and post the full session output;
1. show the log of your `make install` command when you were installing the library - or just the directory listings for me to understand where the compiled module (`_mraa.so`) and the python wrapper (`mraa.py`) are located relative to the script you are trying to run;
1. in the Python shell please run the below commands to see where it actually imports the module from:
```
  >>> import mraa
  >>> print(mraa)
```

v1v3kdalal, 2015-11-02:
1. http://dpaste.com/1GF41AE

2. a. _mraa.so location: /home/johnson/mraa/build/src/python
    b. mraa.py location: /home/johnson/mraa/build/src/python

3. http://dpaste.com/00JC3GD

@alext-mkrs these are results I obtained after following you instructions. Sorry for the late reply as our hard-drive was giving us problems.

Thank you.

alext-mkrs, 2015-11-03:
Ok, looks like there's indeed something wrong with your installation. Can you provide exact steps/commands you've run wjen installing?

I've just installed a Debian 8 VM and was able to successfully build and run mraa on it, though it required a trick of manually copying Python wrapper binaries. Mraa installs packages into Python's site-packages dir, but [Debian's python looks in dist-packages](https://wiki.debian.org/Python#Deviations_from_upstream). But there are different ways of configuring Python and in your case it's importing it but is not able to treat it properly, that's why showing me what exactly you have done will help me to guide you to untangling this :smiley: 

```
tst@deb8-tst:~/mraa/build$ python3
Python 3.4.2 (default, Oct  8 2014, 10:45:20) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
>>> mraa.getVersion()
'v0.8.1'
>>> print(mraa.getVersion())
v0.8.1
```

v1v3kdalal, 2015-11-05:
All of these commands are done in the terminal on our /homedirectory/mraa. I think this is the process we followed:
1. mkdir mraa
2.  apt-get install build-essential python-pip python-dev python-smbus git
3. apt-get install cmake
4. git clone https://github.com/intel-iot-devkit/mraa
5. mkdir build
6. cd build
7. cmake -DBUILDSWIGPYTHON=ON
8.make

I am not sure if we are actually installing after we build the library.
Then we try to use the library by running the tests in /mraa/tests. I.E. general_checks.py, etc.

Thank you for all your help .

alext-mkrs, 2015-11-06:
I see. Well, it's not going to work like that as you're missing some important things. I think while experimenting you might have messed up the environment a little which results in such a strange behavior.

To make it work you need to fully follow the instructons in docs/building.md. Below is a full step-by-step list I ran on my freshly installed Debian 8 VM (this is for Python 2 version):

1. sudo apt-get install git build-essential swig3.0 python-dev cmake
1. git clone https://github.com/intel-iot-devkit/mraa.git
1. cd mraa
1. mkdir build
1. cd build
1. cmake -DSWIG_EXCUTABLE=/usr/bin/swig3.0 -DCMAKE_INSTALL_PREFIX=/usr ..
1. make

After these you can run

1. cd src/python

and you should be able to run `python` being in that dir and do `import mraa` and then `getVersion()`, etc.

Alternatively, if you don't want to depend on being in the same directory as the built mraa module, you can install it globally (which is what I'd recommend). For that run (being in the mraa/build directory)

1. sudo make install
1. sudo ln -s /usr/lib/python2.7/site-packages/* /usr/lib/python2.7/dist-packages

v1v3kdalal, 2015-11-13:
Success!
After following your step by step guide listed in the previous post. 
We are able to get the library working. 
You were right we forgot to install swig; which that makes a lot of sense why all the c code was working and not the python. 

Thank you for your help!

v1v3kdalal, 2015-11-13:
@alext-mkrs @arfoll 
After running all the tests in the mraa/tests folder.
We get an error with the platform_checks.py
In the link is the output of all three tests, could you look at them and tell us if we have another issue?
Thank you.
http://dpaste.com/3NR1XD0

arfoll, 2015-11-14:
The tests are failing because they are made for Edison. To be honest with you the auto tests we have aren't that great, were working on open sourcing a better test framework with a hw harness

alext-mkrs, 2015-11-14:
@v1v3kdalal, glad you have it working now. I will look into adding your suggestions into docs and fixing the need to make symlinks on Debian (they're using a bit nonstandard paths).

v1v3kdalal, 2015-11-16:
@alext-mkrs@arfoll
Are there any python api or documentation which we can use to make our own tests for gpio pins and other checks on the MinnowBoard Max??

Thank you.

arfoll, 2015-11-16:
Documentation for mmax is here: https://github.com/intel-iot-devkit/mraa/blob/master/docs/minnow_max.md

If you look int the tests/ folder you can see how to write your own tests or you can just play witht he gpios directly from python.

v1v3kdalal, 2015-11-17:
@arfoll the tests/folder doesn't have any python api or examples for MinnowBoard Max. I guess it has examples for checks on the Edison board like you mention in your previous post.

It would be great if we have some python api or documentation of functions which help us access GPIO pins on MinnowBoard Max. 

Thank you.

alext-mkrs, 2015-11-18:
Check out the link @arfoll provided in his previous message, it's exactly the Minnow-related documentation you're asking about. General Python API for mraa is described in online docs linkef in the main readme file, check it out.

arfoll, 2015-11-18:
@v1v3kdalal as @alext-mkrs said, there is really no need to have specific minnowboardmax examples, with the docs and general api docs (check py.mraa.io) you'll be able to do everything you need...

commit 90accf7b (2015-11-19): README.md: added install instructions per suggestion in #336.

Closes #336.
commit 63cffab3 (2015-12-04): README.md: added install instructions per suggestion in #336.

Closes #336.
closed 2015-11-19 by alext-mkrs