eclipse-mraa/mraa #337: PWM: mraa limits a period supported by a kernel driver.

xbolshe, 2015-10-25:
mraa has a dedicated limitation of period like b->pwm_max_period definded for a board.
Actually my Galileo Gen1 board supports a period up to 690880 us.
But I cannot select it. I guess only a kernel driver may limit a period, not mraa.

[image]

arfoll, 2015-10-25:
Is this using the official 3.8.x kernel? Weird maybe the limit set in the board config is just wrong, can you verify you can actually set that? If you can I have no issues with raising the max.

xbolshe, 2015-10-25:
I use my custom kernel 3.19.2, where I have changed cy8c9540a driver. And I got 1...690880 us.
Official Intel kernels like 3.8.7 and 3.14.23 for Galileo boards support only a narrow period range like 1...7968 us (do not know why, while cy8c9540a allows to make "690880 us" support easily).

Here is my test:

[image]

[image]

xbolshe, 2015-10-25:
May be it is better to provide information about period limitation in the documentation.

Mark-van-der-Pol, 2017-12-29:
On the MinnowBoard Turbot, the kernel driver through the filesystem sys/class/pwm interface, limits the max period to more than the MRAA Max says.

However, if it is set to more than 5230000, the PWM simply doesn't run.
Also, observing on a scope, the actual period seems to be off by 2x

Looking into this further.

|\\/ark

arfoll, 2017-12-30:
Curious, maybe it's been tweaked again. Typically our max is really the observable max we see on our scopes but on minnowboard I think we trusted what was told to us by our colleagues :)