eclipse-mraa/mraa #338: Provide a meaning to "MRAA"

mulderp, 2015-10-25:
Hi,

as follow up from the discussion on Twitter https://twitter.com/mulpat/status/658258093709635584

It might be nice to get some more meaning into "MRAA".

"Modules" for "wRapping" "hArdware" "Abstractions" maybe?

xbolshe, 2015-10-25:
And why MAA was renamed to MRAA +1:

arfoll, 2015-10-25:
Check #109. The rename from maa was because there was already a libmaa http://sourceforge.net/projects/dict/files/libmaa/libmaa-1.3.2/.

I'll leave this open so you the bike shedding can stay here, but sorry there isn't much reason behind the name... If you buy me a beer I'm happy to have lengthy discussions in persob about my disregard for project naming :)

xbolshe, 2015-10-25:
Hm... MRAA Beer Party.... sounds good :)

mulderp, 2015-10-26:
Interesting to hear the very radical approach to project naming :)

I just know that Arduino was apparently based on the bar where the folks met in Ivrea: https://www.quora.com/Where-is-the-Arduino-bar

For sure, MRAA looks interesting as it provides the needed hardware abstraction layer on somewhat different components with different pins.

mulderp, 2015-10-27:
in the comments on this hackaday article on Arduino is the idea of "HALAL" for Arduino - hardware abstraction layer abstraction layer.... http://hackaday.com/2015/10/20/the-case-for-arduino-in-real-engineering/

rzr, 2016-11-05:
This can be closed : "monkeys really are awesome"

in own @arfoll  words at #IOTWTB 
http://iot.withthebest.com/