eclipse-mraa/mraa #339: Working with JS callbacks

mulderp, 2015-10-25:
Hi,

just curious about why there are no callbacks used in the JavaScript examples.
I am looking at using an input button: When it is pressed, a callback should be executed, e.g.:

      var button = new mraa.Gpio(4);
      button.dir(mraa.DIR_IN);

      button.read(function(d) {
          if (d==1) {
            console.log('ON');
          } else {
            console.log('OFF');
          }
      }, sampleRate);

Would something like this be possible?

mulderp, 2015-10-25:
My current code is:

```
    setInterval(function() {
      var input = button.read();
      console.log(input);
    }, sampleRate);
```

But to detect very fast clicks, sampleRate is e.g. 100ms - so, I was thinking it might be nice to have a callback that is called whenever the button is pressed, without "polling" in setInterval or another kind of loop.

FlippieCoetser, 2015-10-25:
@mulderp  you can configure an Interrupt that has a callback. That way you don't have to loop and read the value continuously. You can further specify when the callback is triggered: On button press or on button depress. Depending on your application you might also have to implement some denouncing hardware or software.

mulderp, 2015-10-25:
Exactly what I am looking for! Thanks! How would I attach/map an interrupt to a Gpio pin?

FlippieCoetser, 2015-10-25:
Their is a function called ISR that you can use to configure the interrupt. Just watch out not all pins can be used with interrupts. Check the documents for more details on the ISR func.

mulderp, 2015-10-25:
right, there is even an example for that:

https://github.com/intel-iot-devkit/mraa/blob/master/examples/javascript/isr.js

thanks!

mulderp, 2015-10-25:
works great:

```
var mraa=require('mraa');

function h() {
  console.log("hello");
}

p = new mraa.Gpio(4);
p.isr(mraa.EDGE_FALLING, h);

setInterval(function() {
  console.log("h");

}, 400);

```

closed 2015-10-25 by mulderp