eclipse-mraa/mraa #341: Python 3 gpio isr method's "args" do not work

hansmosh, 2015-10-27:
I'm using the isr functionality with python 3 and the "args" are not being passed to my callback function properly.
http://iotdk.intel.com/docs/master/mraa/python/mraa.html#mraa.Gpio.isr

Using the example here:
https://github.com/intel-iot-devkit/mraa/blob/master/examples/python/hello_isr.py
If you `print(args)` inside of `test` then it is just a random integer. I want to be able to pass something and then actually use it.

I have tried modifying gpio.c line 243 with some success. What do you think?
```-            arglist = Py_BuildValue("(i)", dev->isr_args);
+            arglist = Py_BuildValue("(O)", dev->isr_args);```

This works with simple things like passing an integer, but I would love to pass `self` and then have my callback function be a method in my class.

arfoll, 2015-10-28:
That would actually be better - do you want to do a PR for this or do you want me to do it?

Passing self should be do-able with this method, or maybe something very close, I'd love to investigate but don't have the time. If someone is willing to spend time doing this I'd be happy to include this.

hansmosh, 2015-10-28:
Copying this over from #343. It mostly fixes this issue but you still can't pass `self` as args. 

```
class Gpio():

  def __init__(self):
    ...
    pin.isr(mraa.EDGE_BOTH, callback, self)

  def callback(self):
    print("we did it")
```

Note: someone can create a new issue for this if they feel it is appropriate to close this one out.