eclipse-mraa/mraa #347: Recent versions of Node.js are not supported

kyrylkov, 2015-10-30:
Building `mraa` on Node.js 4.2.1 fails:

```
/home/root/client/node_modules/mraa/.node-gyp/4.2.1/include/node/node_buffer.h:43:40: note:   candidate expects 3 arguments, 2 provided
mraa.target.mk:122: recipe for target 'Release/obj.target/mraa/src/mraajsJAVASCRIPT_wrap.o' failed
make: *** [Release/obj.target/mraa/src/mraajsJAVASCRIPT_wrap.o] Error 1
make: Leaving directory '/home/root/client/node_modules/mraa/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/opt/node-v4.2.1-linux-x86/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 3.10.17-poky-edison+
gyp ERR! command "/opt/node-v4.2.1-linux-x86/bin/node" "/opt/node-v4.2.1-linux-x86/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/root/client/node_modules/mraa
gyp ERR! node -v v4.2.1
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok
npm ERR! Linux 3.10.17-poky-edison+
npm ERR! argv "/opt/node-v4.2.1-linux-x86/bin/node" "/opt/node-v4.2.1-linux-x86/bin/npm" "install" "mraa"
npm ERR! node v4.2.1
npm ERR! npm  v2.14.7
npm ERR! code ELIFECYCLE

npm ERR! mraa@0.8.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the mraa@0.8.0 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/root/client/npm-debug.log
```

xbolshe, 2015-10-31:
See https://github.com/intel-iot-devkit/mraa/issues/335

alext-mkrs, 2015-11-03:
@kyrylkov try again - after 0.8.1 was released by @tingleby you should be good. I see npm already has it.

kyrylkov, 2015-11-03:
@alext-mkrs Yes thanks. I will try soon.

closed 2015-11-03 by kyrylkov