eclipse-mraa/mraa #349: Install use Windows 8.1

mtsys, 2015-11-02:
Already followed other tutorials and nothing able to install the lib mraa. I am use Windows 8.1

C:\>npm install mraa
|
> mraa@0.8.1 install C:\node_modules\mraa
> node-gyp rebuild

C:\node_modules\mraa>if not defined npm_config_node_gyp (node "C:\Program Files\
nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-g
yp.js" rebuild )  else (node  rebuild )
'sed' não é reconhecido como um comando interno
ou externo, um programa operável ou um arquivo em lotes.
gyp: Call to 'node -e 'console.log(process.versions.v8)' | sed 's/\.//g'' return
ed exit status 255. while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (C:\Program Files\nodejs\node_module
s\npm\node_modules\node-gyp\lib\configure.js:355:16)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_proces
s.js:200:12)
gyp ERR! System Windows_NT 6.3.9600
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodej
s\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\node_modules\mraa
gyp ERR! node -v v4.2.1
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok
npm ERR! Windows_NT 6.3.9600
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\
node_modules\\npm\\bin\\npm-cli.js" "install" "mraa"
npm ERR! node v4.2.1
npm ERR! npm  v2.14.7
npm ERR! code ELIFECYCLE

npm ERR! mraa@0.8.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the mraa@0.8.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\npm-debug.log

arfoll, 2015-11-02:
Sorry but you can't compile mraa on windows, this implementation is Linux only. What are you trying to achieve? Maybe there's another way

mtsys, 2015-11-02:
I was seeing the ocean to attend an Intel event on Edsion. I thought I could use it on Windows for testing. As my plate has not yet come, he was only studying.

closed 2015-11-02 by mtsys

arfoll, 2015-11-02:
Fair enough, I'd recommend just using a Linux VM, mraa compiles well on
Ubuntu/fedora/Arch/Debian and you'll be ready for Edison!